cantools/cantools #4: pyparsing.ParseException: Expected end of text

slovak194, 2017-07-07:
The newest version of a cantools (4.0.0) crashes while parsing dbc on string:
BA_ "BusType" "CAN";

Error log:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/cantools/db.py", line 547, in add_dbc
    tokens = self._grammar.parseString(dbc.read())
  File "/usr/local/lib/python3.5/dist-packages/pyparsing.py", line 1632, in parseString
    raise exc
  File "/usr/local/lib/python3.5/dist-packages/pyparsing.py", line 1622, in parseString
    loc, tokens = self._parse( instring, 0 )
  File "/usr/local/lib/python3.5/dist-packages/pyparsing.py", line 1379, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/usr/local/lib/python3.5/dist-packages/pyparsing.py", line 3395, in parseImpl
    loc, exprtokens = e._parse( instring, loc, doActions )
  File "/usr/local/lib/python3.5/dist-packages/pyparsing.py", line 1383, in _parseNoCache
    loc,tokens = self.parseImpl( instring, preloc, doActions )
  File "/usr/local/lib/python3.5/dist-packages/pyparsing.py", line 3183, in parseImpl
    raise ParseException(instring, loc, self.errmsg, self)
pyparsing.ParseException: Expected end of text (at char 10209), (line:218, col:1)
```

File fragment:
```
BA_DEF_DEF_  "BusType" "CAN";
BA_ "BusType" "CAN";
BA_ "GenMsgCycleTime" BO_ 13 1;
```

Old, 1.0.0, works as expected.

eerimoq, 2017-07-07:
The new version raises an exception if the parser fails to parse the whole file, while the old version silently ignored the rest of the file. Also, a lot of new syntax is accepted, ehich may have introduced a bug or two. BA_ and BA_DEF_DEF_ were introduced by pull request https://github.com/eerimoq/cantools/pull/2 recently, and not tested at all by me. No unit tests exists.

There are two options:

1. As a workaround, delete rows in the DBC-file that fails.
2. Fix the bug in cantools/db.py.

I can have a look tomorrow, but would be nice if you can make a fix and submit a pull request (with unit tests) =)

eerimoq, 2017-07-07:
Ok, I craeted a new release, 4.0.1, which successfully parses your failed DBC-lines. Let me know if there are more problems, likely there is since the parser is not yet accepting all DBC-file syntax.

eerimoq, 2017-07-12:
Just curious, did it solve your parsing problem?

eerimoq, 2017-07-16:
Closing since no response was received, and that usually means it works.

closed 2017-07-16 by eerimoq

slovak194, 2017-12-10:
Sorry, I was not able to answer your questions.
In the nearest future, I'll possibly again work with your amazing library and give you more feedback.
Best regards,
Alex