cantools/cantools #5: dbc paring, message._fmt with wrong padding

karlpeterhugo, 2017-08-10:
Hi, first cool project!

I think there is a bug in computing the padding for some messages:

I have a messgae format in DBC like this:
```
BO_ 851 MSG0: 8 GW
 SG_ A : 6|15@0+ (0.015625,0) [0|511.984] "x"  ECU
 SG_ B : 7|1@0+ (1,0) [0|1] ""  ECU
 SG_ C : 38|15@0+ (0.015625,0) [0|511.984] ""  ECU
 SG_ D : 39|1@0+ (1,0) [0|1] ""  ECU
```

It results in a fmt string with more than 64 bit due to too large padding like this:
`p24u2u15p30u1u15`

I don't know the exact syntax of the DBC file, but the definition of start in not intuitive.
The Vector DBC editor reports as startbits:
D->39
C->40
B->7
A->8

I think the logic in this function of start + length may not be right. May be even the start bit coputed by your code before is wrong.

```
def _create_message_encode_decode_format(signals):
    fmt = ''
    end = 64
    for signal in signals:
        padding = end - (signal.start + [signal.length)
```

Let me know it this it the right way to report the issue, if you plan to fix it or would like to have more information.

eerimoq, 2017-08-10:
Hello,

this is the right place to file issues. I don't know all details about the DBC format myself, and I'm happy no pointed out that there is a problem in the package.

The ``0`` after the ``@`` is the byte order of the signal. Looks like big endian byte order format is incorrect indeed, the number of bits in the format should be max 64.

The problem is probably not in ``_create_message_encode_decode_format``, but that the start bit in big endian signals should be converted somehow to the internal represenation of the frame data. I don't quite understand how to perform this conversion, and I don't have any good tools available as guidance, so it would be lovely if you can provide more information about the expected format string.

I made some [changes on a branch](https://github.com/eerimoq/cantools/commit/bd90c6052efb6600736ecc9526814582c13dff95) that gives 64 bits in the format string (``u16u15u1u16u15u1``), but I'm not sure it is correct. B is mapped to bit 0, and A to bit 1, which does not match your Vector output. Anyway, the problem you highlight in this issue should be fixed, we just have to figure out how.

karlpeterhugo, 2017-08-10:
Hi,
thank you for the quick response. I will report if it works tomorrow.

> I made some changes on a branch <https://github.com/eerimoq/cantools/commit/bd90c6052efb6600736ecc9526814582c13dff95> that gives 64 bits in the format string (u16u15u1u16u15u1), but I'm not sure it is correct. B is mapped to bit 0, and A to bit 1, which does not match your Vector output.

That might be quite right. The Vector Editor also shows a bit-layout that matches your format string (2 bytes are used, 2 unused, 2 used, 2 unused). I think its just Vectors is counting bits inside a byte the other way around, hence showing non intuitive values for their signal.start.

karlpeterhugo, 2017-08-11:
Hi,
not the fields length seems right.
But the interpretation of the bit-order of the signals A and C is wrong.

Attached is a view from the layout editor.
As said, vectors bit counting is unusual and so there is a second picture with bit index inverted which seems to be the natural counting for me:
-signal B is bit 0 of byte 0
-signal A is from bit 1 of byte 0 to bit 15 of byte 1
-concerning the bit order of A, bit 1 of byte0 is the MSB (even if it is called Motorola byte order).

Does that make sense?

[image]

eerimoq, 2017-08-11:
I made [another attempt on the same branch](https://github.com/eerimoq/cantools/commit/edc4c4744733d6be4862317287195240da90dddc) with the bit ordering reversed in big endian signals, but it's not clear to me if it is correct. I can imagine it's not as simple as to reverse the bit order, but probably depends on byte boundaries as well.

Can you provide the frame data sent by the Vector tool, and its decoded signal values?

karlpeterhugo, 2017-08-13:
Hi, first simple test looks good. I plan to confirm and provide a test case on Monday.

karlpeterhugo, 2017-08-14:
Hi,
unfortunately it appears to still not match the Vector format. I added a test case in a fork, but did not change the implementation.
Commit 5f14cff42864b86798d2518297f8af075b588779

I'm still new to github. This is the way to add test code from my side, right?

eerimoq, 2017-08-14:
Yeah, a fork is usually how you share code.

I changed the code to make your test pass. The rest of the tests fails, but that's fine for now. Let me know if the fix solves your problem. The bit/byte ordering was pretty much a guess from my side from the beginning, so it wouldn't surprise me if all signals were mixed up in the currently release version. So any changes we make are probably to the better.

Will be very nice when the logic is correct for all configurations. =)

karlpeterhugo, 2017-08-15:
Hi,
now my signals, including signals of several other messages, are decoded correctly :-)

So which type of message signals are now expected to fail?
Most of my signals are of form @+0 and those appear to pass.

eerimoq, 2017-08-15:
Nice, Can you add another testcase, similar to what you already added, to make me even more confident I don't break anything in the future?

I'm not sure actually. I expect @1 to fail since I have no reference data of that kind, and usually it's hard to make a correct implementation by guessing and misinterpreting specifications. =)

I found https://se.mathworks.com/help/vnt/ug/canpack.html, maybe it can help me out. Looks neat, but it does not contain reference DBC file and CAN frames. =/

eerimoq, 2017-08-15:
Did additional changes to make little endian byte order work (at least I think so). The big endian byte order still pass, but encoded frame data is padded with two zero bytes to a total of 8 bytes because it simplified the implementation. Cannot tell if padding is expected by the user, so I'll leave it like this until someone have a good argument for changing it.

https://github.com/eerimoq/cantools/commits/issue-5

eerimoq, 2017-08-16:
I created a new release, 9.0.0, and uploaded it to PyPi.

Thanks for the help, and let me know if there are more problems. Pull requests are very welcome!

eerimoq, 2017-08-18:
Reopen this issue or create a new one if the issue is still present. Thanks for the help!

closed 2017-08-18 by eerimoq

karlpeterhugo, 2017-08-19:
Hi, issue appears to be solved. With 9.0.0 I did not see any problem in a dbc with various signals of different byte order. Thank you for fixing!

eerimoq, 2017-08-19:
No problem, always satisfying improving the code making users happy =)