cantools/cantools #6: feature request: message.decode(data, decode_choices=False)

karlpeterhugo, 2017-08-30:
Hi,
I suggest to support a parameter to disable the decoding of signal choices (e.g. enumerated signal values such as 'ON', 'OFF') and to decode the integer value instead.
This can be useful for post processing.

eerimoq, 2017-08-30:
Sounds great, easy to implement and useful. Would be nice if you could implement it and submit a pull request.

eerimoq, 2017-08-30:
At the same time we could make scaling optional, an there sometimes are rounding errors.

eerimoq, 2017-09-01:
Your suggestions has been implemented and released in 9.1.0!

closed 2017-09-01 by eerimoq

karlpeterhugo, 2017-09-06:
Hi Erik, the new feature is great and helps me using your cantools package out of the box.
Thank you for implementing it!

eerimoq, 2017-09-06:
You're welcome :)