cantools/cantools #7: Message.nodes returns string

zlim00, 2017-09-10:
Message.nodes returns a string not a list as the documentation suggests

eerimoq, 2017-09-10:
Please give an example showing your input and output, and I can create a unit test using that data, and then make nodes a list instead of a string.

zlim00, 2017-09-10:
```
database = cantools.db.load_file("database.dbc")
for message in database.messages:
        print repr(message.nodes)
```
output are strings, method documentation states output should be a lists

eerimoq, 2017-09-10:
Fixed on master, please give it a try and I'll release it to PyPi if you are satisfied.

zlim00, 2017-09-10:
seems to work 👍 
Related question, how is multiple nodes for a message defined in the database? (can't find a specification and can't test with candb++)

eerimoq, 2017-09-10:
I don't know if the DBC format can have multiple nodes (or producers) per message. Since this package is intended to support multiple database formats sharing a common API I decided to make message nodes a list as some format may allow multiple nodes per message. Good or bad decision is for the future to decide. =)

eerimoq, 2017-09-10:
I uploaded a new release to PyPi, 10.0.0. Let's hope not too many users relied on the nodes as a string. Anyway, it's very easy to port the code to the new version by adding `[0]` after the nodes member.

closed 2017-09-10 by eerimoq

artemiljin, 2019-01-30:
Hi,
  DBC format supports multiple nodes associated with different ECUs, it'd be nice to have merging of nodes, ecus in add_dbc_file method.

eerimoq, 2019-01-30:
Hello,

feel free to implement it. =)

artemiljin, 2019-01-30:
> Hello,
> 
> feel free to implement it. =)

Nice, just wanted to be sure that it wasn't planned.