cantools/cantools #8: Messages always encoded with 8 bytes

AxelVoitier, 2017-09-25:
Hello,

Thanks for putting this nice library out!

I found what I think is a defect: when I encode a message, it is always returned on full length of 8 bytes, even though a smaller length is correctly set in the description file (Kayak in my case).

Steps to reproduce:
1- Have a message of total length 1 byte defined in a .kcd file. Example (just the message itself):
```
<Message id="0x275" length="1" name="Command">
  <Notes>A command</Notes>
  <Signal name="Start" offset="0" length="8" endianess="little" />
  <Producer>
    <NodeRef id="0"/>
  </Producer>
</Message>
```
2- Encode it:
```
import cantools
db = cantools.db.load_file('Test_file.kcd')
messages = {msg.frame_id: msg for msg in db.messages}
messages[0x275]
message('Command', 0x275, False, 8, 'A command')
data = messages[0x275].encode({'Start': 1})
print(len(data))
# Printing: 8
```

Not only does it returns 8 bytes long data. The Message object itself believe it is of length 8.

I am willing to find the source of the problem and submit a PR. But before, I wanted to know if this was an intentional feature or if this is a bug?

Cheers,
Axel

AxelVoitier, 2017-09-25:
OK, length is just hardcoded to 8 in the formats/kcd.py file :D.
And it seems properly set in the dbc format.

I noticed others defect that are just coming from incomplete or incorrect parsing of the kcd format. I can fix most of them. But should I make one PR per issue? Or I do all the modifications (with separate commits at least) and just do one PR for all?

AxelVoitier, 2017-09-25:
Apparently, even when the Message object has the proper length (with dbc, or with fixed kcd parser), it stills produce 8 bytes long data.

With the proper length on the Message object I can easily chop it on the user side:
```
data = messages[0x275].encode({'Start': 1})
length = messages[0x275].length
data = data[:length]
```

But why would the lib not just do it?

eerimoq, 2017-09-25:
Hello,

pull request(s) with improvements of the KCD parser and message encoding that returns the correct number of bytes instead of 8 is very welcome. I can't remember why encoded messages are always 8 bytes. There is probably no good reason, but I might be wrong.

commit 376b250a (2017-09-26): Fixes #8: Messages always encoded with 8 bytes.
commit 931db0ba (2017-09-26): Merge pull request #9 from AxelVoitier/master

Fixes #8: Messages always encoded with 8 bytes.
closed 2017-09-26 by eerimoq