cantools/cantools #13: Issues with comments

xatavian, 2017-11-30:
Hello,

I have a dbc file which cannot be parsed by `cantools`. I looked at `pyparsing` Exception and it looks like the problem comes from one of the comments. 

The error can be seen in the following dbc string:

```
VERSION "TEST"
CM_ "";
```

This crashes and `pyparsing` outputs `Expected end of text (at char 15), (line:2, col:1)`

I also tried with : 
```
VERSION "TEST"
BS_: 
```
And it worked fine.

Best regards,

eerimoq, 2017-11-30:
Hi,

any possibility of posting the entire DBC-file? It would be very helpful to test a fix.

xatavian, 2017-12-03:
Hi,

I'm sorry but i'm not allowed to share the dbc file. I can try to make a pull request but i don't know how pyparsing works so i'd havet to take a look at that :) !

Best regards,

eerimoq, 2017-12-03:
I implemented a fix in 1d6247d9dc1a8b5b8fbc9d2b16be79aa7f37a676.

Let me know if it works or not.

vargabalint92, 2017-12-06:
Hi

unfortunately i have still some problems with comments.

eerimoq, 2017-12-06:
Closing issue since a detailed description of the problem is missing.

closed 2017-12-06 by eerimoq