cantools/cantools #14: Attributes and enums

sbrw, 2017-12-10:
First of all, great tool and nice coding!

A question regarding attributes of enum-type:
Examining DBC-files generated by Vectors tool "CANdb++" (version 3.5.4), I see that the attribute default is defined with the enum-name as a string, but each object is defined with the enumerated value as integer (0 => "cyclic" in example below), not a string.

```
BA_DEF_ BO_  "GenMsgSendType" ENUM  "cyclic","NotUsed","IfActive","NotUsed","NotUsed","NotUsed","NotUsed","NotUsed","noMsgSendType";

BA_DEF_DEF_  "GenMsgSendType" "noMsgSendType";

BA_ "GenMsgSendType" BO_ 2348875518 0;
```
It seems as if cantools always applies the enum-name-string in all cases?
Which DBC-version do you support?

Thanks!

eerimoq, 2017-12-11:
Hello,

the implementation does not follow any specific DBC-version, as no specification is available to me. Instead, it's based on a few DBC-files found on the internet.

The enum handling looks like a bug. The goal is to make the tool compatible with CANdb++ (and other tools implementing the DBC-specification), so any inconsistencies should be fixed. Pull requests are welcome.

/ Erik

eerimoq, 2017-12-16:
Should I close this issue, or wait for a pull request with fixes?

sbrw, 2017-12-17:
Hi again!
I have a pending fix in my fork, but realized it was not complete. I will attend to it as soon as I have time...
/Sven

eerimoq, 2018-01-03:
Should I leave it open?

sbrw, 2018-01-03:
Sorry for the delay. Haven't had the time to fix it yet.

Close the issue if you find it appropriate. I intend to fix it and provide a pull request anyhow...if you see fit.

eerimoq, 2018-01-03:
Let's leave it open then so others can see that there is someone intending to provide a fix.

eerimoq, 2018-01-18:
Closing this issue for now since there is no progress and it is probably not needed by many users.

closed 2018-01-18 by eerimoq