cantools/cantools #17: Cannot decode FLOAT format

afherlam, 2017-12-13:
Hi,

I tried to decode my .dbc file using this tool and when I loaded the .dbc file, I got this following error 

cantools.db.UnsupportedDatabaseFormatError: DBC: "Invalid DBC syntax at line 2724, column 33: 'BA_DEF_DEF_  "SamplePointMax" 81>!<.5;': Expected ";"."

After spending a little bit of time to find the error, it turned out, that this error is related to the following signal which has FLOAT data type
     BA_DEF_  "SamplePointMax" **FLOAT** 0 100;

I really appreciate your help. 

Thanks in advance.

eerimoq, 2017-12-13:
Hello,

it's not related to `BA_DEF_ "SamplePointMax" FLOAT 0 100;`, but to the fact that the parser does not allow a floating point number in `BA_DEF_DEF_`.

I'll have a look at it, should be easy to implement a fix.

eerimoq, 2017-12-13:
Recently released version 14.3.0 can successfully parse the floating point number above.

closed 2017-12-13 by eerimoq

afherlam, 2017-12-14:
Hello,
thanks for you fast release and bug fix.

But I still face the same problem with another attribute e.g. BA_, so I got the following failure:

> cantools.db.UnsupportedDatabaseFormatError: DBC: "Invalid DBC syntax at line 2758, column 24: 'BA_ "SamplePointMax" 81>!<.25;': Expected ";"."

Thanks again in advance.

eerimoq, 2017-12-14:
Test 14.3.1.

afherlam, 2017-12-17:
Hi,
thanks for your help, you are very quick.
Now it works fine :)

One thing from my side (I hope this is the last one ;)):
I assume your tool only supports utf-8 file format, because when i tried to load a .dbc file with ANSI format, I got the following error message:

> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb2 in position 1861: invalid start byte

Is there a way to read the ANSI file format? Because right now I do the workaround by changing the file format first before I load the file with your tool :)

Many thanks again in advance

eerimoq, 2017-12-17:
I don't think there is, no. What would be needed to support other character encodings?

afherlam, 2017-12-17:
The traceback looks like this:

>    db = cantools.db.load_file("data/example.dbc")
>   File "/usr/local/lib/python3.4/dist-packages/cantools/db/__init__.py", line 41, in load_file
>     return load(fin)
>   File "/usr/local/lib/python3.4/dist-packages/cantools/db/__init__.py", line 58, in load
>     return load_string(fp.read())
>   File "/usr/lib/python3.4/codecs.py", line 319, in decode
>     (result, consumed) = self._buffer_decode(data, self.errors, final)
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb2 in position 1861: invalid start byte

My workaround looks like this:
```
dbc_file = open("data/example.dbc", "r", encoding="iso-8859-1")
db = cantools.db.load(dbc_file)
```
So first I open the file using the correct encoding (for this i need to know what kind of encoding the file has), and then load the object with the command cantools.db.load

blackskandix, 2017-12-17:
> My workaround looks like this:
> ```
> dbc_file = open("data/example.dbc", "r", encoding="iso-8859-1")
> db = cantools.db.load(dbc_file)
> ```
That's is already the solution - I'm using the same for my "shift_jis" encoded files.

BTW: inside cantools, the solution would be the same line of code, 
since `load_file()` just opens a filepointer with `open`
```
def cantools.db.load_file(filename):
  with open(filename, 'r') as fin:
    return cantools.db.load(fin)
```

eerimoq, 2017-12-17:
I added the `encoding` argument to `load_file()`, but doing so requires `from io import open` in Python 2, and then other Python 2 and 3 portability issues are introduced. I'll leave this for now as you have a workaround for your case, and I would appreciate help implementing character encoding properly, both on Python 2 and 3 (only Python 2 is problematic I think). I guess unicode should be used instead of strings in Python 2, but I don't know...

An alternative is to only allow character encoding in Python 3.