cantools/cantools #19: SG_MUL_VAL_

emmhaych92, 2018-01-04:
I am having issues parsing SG_MUL_VAL_ parameter in .dbc files. The following exception is raised:

```
>>> db = cantools.db.load_file('TestDB.DBC')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\cantools\db\__init__.py", line 41, in load
_file
    return load(fin)
  File "C:\Python27\lib\site-packages\cantools\db\__init__.py", line 58, in load

    return load_string(fp.read())
  File "C:\Python27\lib\site-packages\cantools\db\__init__.py", line 88, in load
_string
    raise UnsupportedDatabaseFormatError(e_dbc, e_kcd)
cantools.db.UnsupportedDatabaseFormatError: DBC: "Invalid DBC syntax at line 77,
 column 1: '>!<SG_MUL_VAL_ 2148676694 BIT_L Multiplexor 8-8, 16-16, 24-24;': Exp
ected end of text.", KCD: "syntax error: line 1, column 0"
```

The SG_MUL_VAL_ defines the multiplexer for the signal. For example:
`SG_MUL_VAL_ 2148676694 BIT_L Multiplexor 8-8, 16-16, 24-24;`
- 2148676694: Message ID
- BIT_L: Signal to multiplex
- Multiplexor: Multiplexer field
- 8-8, 16-16, 24-24: Values to multiplex

Here is the dbc: [TestDB.zip](https://github.com/eerimoq/cantools/files/1602718/TestDB.zip)
dbc is generated using CANdb++

eerimoq, 2018-01-04:
I implemented what I think is a fix in 1296bda0bfdd5e0c7945be57d8f26dc0f7609499.

This only adds support for SG_MUL_VAL_ in the parser. The parsed value it not used at all as I don't know how to use it.

Let me know if the issue persists.

eerimoq, 2018-01-05:
Did it work?

emmhaych92, 2018-01-05:
Thanks for the prompt response. https://github.com/eerimoq/cantools/commit/1296bda0bfdd5e0c7945be57d8f26dc0f7609499 allows me to import the database without throwing the exception - I have tested this on several of our full scale databases too and it works!

In regards to making this parameter functional, Vector has a document explaining how extended multiplexing works: https://vector.com/portal/medien/cmc/application_notes/AN-ION-1-0521_Extended_Multiplexing_in_DBC_Databases.pdf. Figures 1 and 2 in this document probably illustrate the functionality we want i.e. given the multiplexer value, different signals exist in the message.

The [TestDB.dbc](cantools/tests/files/multiplex.dbc) i provided you earlier consists of a message `Message1` which has a CAN ID of 0x123456. It's first signal is `Multiplexor` on bit locations 2:7 which is used to determine what signals exist in the message.
- If `Multiplexor = 0x8`: Message consists BIT_G, BIT_C, BIT_J, BIT_L signals 
[image]
- If `Multiplexor = 0x10`: Message consists of BIT_G, BIT_C, BIT_J, BIT_L signals (same as before)
[image]
- If `Multiplexor = 0x18`: Message consists of a lot more signals (BIT_G, BIT_C, BIT_J, BIT_E, BIT_K, BIT_A, BIT_L, BIT_F, BIT_H, BIT_B, BIT_D)
[image]

This is what the library currently does:
```
>>> db.decode_message(0x123456, b'\x20\xFF\xFF\xFF\xFF\xFF\xFF\xFF') # Multiplexor 0x08
{'Multiplexor': 8, 'BIT_C': 1, 'BIT_J': 1, 'BIT_L': 1, 'BIT_G': 1}
>>> db.decode_message(0x123456, b'\x60\xFF\xFF\xFF\xFF\xFF\xFF\xFF') # Multiplexor 0x18
{'Multiplexor': 24, 'BIT_H': 1, 'BIT_K': 1, 'BIT_A': 1, 'BIT_B': 1, 'BIT_E': 1, 'BIT_D': 1, 'BIT_F': 1}
```
As you can see when multiplexor is 8 (0x08), the correct signals are decoded. However, when the multiplexor is 24 (0x18) the signals `BIT_C`, `BIT_J`, `BIT_L` and `BIT_G` are missing.

In the DBC file:
```
SG_ BIT_L m8 : 24|1@1+ (1,0) [0|0] "" Vector__XXX
.
.
SG_MUL_VAL_ 2148676694 BIT_L Multiplexor 8-8, 16-16, 24-24;
```
The m8 needs enables BIT_L to work when multiplex is 8. The `SG_MUL_VAL_` line indicates that for `BIT_L` m8, m16 and m24 are valid multiplexer.

eerimoq, 2018-01-05:
Hello,

give 57e7cfe9da509d184f91937baf390e6d9847f49d a try. It successfully encodes and decodes Message1 with Multiplexor 8, 16 and 24.

Here is the encoding and decoding test: https://github.com/eerimoq/cantools/blob/57e7cfe9da509d184f91937baf390e6d9847f49d/tests/test_cantools.py#L978-L1009

eerimoq, 2018-01-05:
There was a few errors in recent commits, but the latest commit on master should work. Please test using latest commit on master instead of any other commit. =)

emmhaych92, 2018-01-08:
Thanks! I tested using 6b945201512495e9b64bad8497461c719376a3d7.

Testing with our main database, I am having an issue with [message.py Line 383](https://github.com/eerimoq/cantools/blob/master/cantools/db/message.py#L383). I've replicated the cause if this in: [TestDB_Multiplex_Value_Desc.zip](https://github.com/eerimoq/cantools/files/1611377/TestDB_Multiplex_Value_Desc.zip)

**Description of Issue**

The database I provided you initially was using `Multiplexor` values 8, 16 and 24. It did not consist of value descriptors; mapping values of a signal to descriptive text. 

If a value descriptor exists for the signal e.g. `8 = MULTIPLEX_8`, the [message.py Line 383](https://github.com/eerimoq/cantools/blob/master/cantools/db/message.py#L383) searches for `MULTIPLEX_8` in `multiplexers[signal]` whilst the dictionary's key is 8.

This issue can be replicated with [TestDB_Multiplex_Value_Desc.zip](https://github.com/eerimoq/cantools/files/1611377/TestDB_Multiplex_Value_Desc.zip) as follows:

```
>>> db.decode_message(0x123456, b'\x20\xFF\xFF\xFF\xFF\xFF\xFF\xFF')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "cantools\db\file.py", line 325, in decode_message
    return message.decode(data, decode_choices, scaling)
  File "cantools\db\message.py", line 417, in decode
    return self._decode(self._codecs, data, decode_choices, scaling)
  File "cantools\db\message.py", line 393, in _decode
    node = multiplexers[signal][mux]
KeyError: 'MULTIPLEXOR_8'
```
If you inspect `multiplexers[signal]` you'll see the keys are 8, 16 and 24.

Apart from this I think it is working as expected!

eerimoq, 2018-01-08:
Fixed and part of release 15.0.0 on PyPI.

NOTE: Release 15.0.0 requires bitstruct 3.7.0, so you might have to upgrade that package manually (`pip install -U bitstruct`).

closed 2018-01-09 by eerimoq