cantools/cantools #20: Are DBC signals of value type "IEEE Float" supported?

karlpeterhugo, 2018-01-04:
Hi,
I have a dbc file which defined some signals as 32bit IEEE float. I think in DBC file this is done by an extra line, but I'm unsure.

`SIG_VALTYPE <can-id> <signame> : 1;`

With cantools 9.1.1 I got parsing errors.
With cantools 14.4.0 the dbc is accepted but the decoded signals looks like to be interpreted as INT32 and not as float.

Thank you for any hints.

eerimoq, 2018-01-04:
I implemented the requested functionality on master, and more specifically in this commit: 6ff17e80d1f8546eb8402113de6ac31c1fed0d92

Give it a try and let me know the results.

karlpeterhugo, 2018-01-04:
Erik you are awesome! First very brief try seems to show reasonable results.  I cannot do more tests before next week, but first tests look correct.

eerimoq, 2018-01-04:
Great, I'll create a new release when you finished your testing.

eerimoq, 2018-01-05:
Would be nice it you test the latest commit on master. Quite a few changes recently and they will all be part of the next release.

karlpeterhugo, 2018-01-08:
I tested commit 6ff17e80d1f85 and all floating point signals work as expected. In addition, my non-floating point signals appear unaffected and correct.

eerimoq, 2018-01-08:
Fixed and part of release 15.0.0 on PyPI.

NOTE: Release 15.0.0 requires bitstruct 3.7.0, so you might have to upgrade that package manually (`pip install -U bitstruct`).

closed 2018-01-09 by eerimoq