cantools/cantools #21: Error when trying to load dbc file: "AttributeError: no such attribute name"

sietschie, 2018-02-07:
I am using the version 16.2.0. When I try to parse a dbc file I get the following error message:

`AttributeError: no such attribute name`

I can reproduce the error with the following line of code:

```
import cantools
cantools.db.formats.dbc.load_string("")
```

The complete backtrace is:

> Python 2.7.13 |Continuum Analytics, Inc.| (default, Dec 20 2016, 23:09:15)
> Type "copyright", "credits" or "license" for more information.
> 
> IPython 5.3.0 -- An enhanced Interactive Python.
> ?         -> Introduction and overview of IPython's features.
> %quickref -> Quick reference.
> help      -> Python's own help system.
> object?   -> Details about 'object', use 'object??' for extra details.
> 
> In [1]: import cantools
> In [2]: cantools.db.formats.dbc.load_string("")
> ---------------------------------------------------------------------------
> AttributeError                            Traceback (most recent call last)
> <ipython-input-2-c260aaeb5b25> in <module>()
> ----> 1 cantools.db.formats.dbc.load_string("")
> 
> /lib/python2.7/site-packages/cantools/db/formats/dbc.pyc in load_string(string)
>     867     """
>     868
> --> 869     grammar = _create_grammar()
>     870
>     871     try:
> 
> /lib/python2.7/site-packages/cantools/db/formats/dbc.pyc in _create_grammar()
>     273                                                              - Suppress('-')
>     274                                                              - Suppress(positive_integer)))
> --> 275                                        - scolon)
>     276     signal_multiplexer_values.setName(SIGNAL_MULTIPLEXER_VALUES)
>     277
> 
> /lib/python2.7/site-packages/pyparsing.pyc in delimitedList(expr, delim, combine)
>    3129        as a list of tokens, with the delimiters suppressed.
>    3130     """
> -> 3131     dlName = _ustr(expr)+" ["+_ustr(delim)+" "+_ustr(expr)+"]..."
>    3132     if combine:
>    3133         return Combine( expr + ZeroOrMore( delim + expr ) ).setName(dlName)
> 
> /lib/python2.7/site-packages/pyparsing.pyc in _ustr(obj)
>     120             # If this works, then _ustr(obj) has the same behaviour as str(obj), so
>     121             # it won't break any existing code.
> --> 122             return str(obj)
>     123
>     124         except UnicodeEncodeError:
> 
> /lib/python2.7/site-packages/pyparsing.pyc in __str__(self)
>    2388
>    2389         if self.strRepr is None:
> -> 2390             self.strRepr = "{" + " ".join( [ _ustr(e) for e in self.exprs ] ) + "}"
>    2391
>    2392         return self.strRepr
> 
> /lib/python2.7/site-packages/pyparsing.pyc in _ustr(obj)
>     120             # If this works, then _ustr(obj) has the same behaviour as str(obj), so
>     121             # it won't break any existing code.
> --> 122             return str(obj)
>     123
>     124         except UnicodeEncodeError:
> 
> /lib/python2.7/site-packages/pyparsing.pyc in __str__(self)
>    2388
>    2389         if self.strRepr is None:
> -> 2390             self.strRepr = "{" + " ".join( [ _ustr(e) for e in self.exprs ] ) + "}"
>    2391
>    2392         return self.strRepr
> 
> /lib/python2.7/site-packages/pyparsing.pyc in _ustr(obj)
>     120             # If this works, then _ustr(obj) has the same behaviour as str(obj), so
>     121             # it won't break any existing code.
> --> 122             return str(obj)
>     123
>     124         except UnicodeEncodeError:
> 
> /lib/python2.7/site-packages/pyparsing.pyc in __str__(self)
>    1379
>    1380     def __str__( self ):
> -> 1381         return self.name
>    1382
>    1383     def __repr__( self ):
> 
> /lib/python2.7/site-packages/pyparsing.pyc in __getattr__(self, aname)
>    1422             return ret;
>    1423         else:
> -> 1424             raise AttributeError("no such attribute " + aname)
>    1425
>    1426     def __eq__(self,other):
> 
> AttributeError: no such attribute name
>

eerimoq, 2018-02-07:
I get this output for the same input:

``` python
> python 
Python 2.7.12 (default, Nov 20 2017, 18:23:56) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import cantools
>>> import pyparsing
>>> cantools.__version__
16.2.0
>>> pyparsing.__version__
2.2.0
>>> cantools.db.formats.dbc.load_string("")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "cantools/db/formats/dbc.py", line 879, in load_string
    e.msg))
cantools.db.formats.utils.ParseError: Invalid DBC syntax at line 1, column 1: '>!<': Expected {VERSION | NS_ | BS_ | BU_ | BO_ | CM_ | BA_DEF_ | BA_DEF_DEF_ | BA_ | VAL_ | VAL_TABLE_ | SIG_VALTYPE_ | SG_MUL_VAL_ | BO_TX_BU_ | BA_DEF_REL_ | BA_DEF_DEF_REL_ | BA_REL_ | EV_}.
>>> 
```

Is there a problem with your Python or pyparsing installation? I'm using pyparsing 2.2.0.

sietschie, 2018-02-07:
I was using pyparsing 1.5.6. After I manually installed version 2.2.0 I got the same message you got and the DBC file gets parsed as expected. Thank you. Maybe you can add the minimum required version of pyparsing to the requirements of the tool so that the setup makes sure the correct version is installed?

And thank you for the quick reply.

eerimoq, 2018-02-07:
There is already an requirement for pyparsing 2.0.3: https://github.com/eerimoq/cantools/blob/master/setup.py#L21-L22

Not sure why your pyparsing was not updated, maybe one have to do `pip install -U cantools`, to force packages to be updated? I'm a setuptools novice, so I might be wrong.

sietschie, 2018-02-07:
Okay. This probably was an issue with my python. For some reason pip thought, 2.2.0 was installed even though it wasn't. Thank you for your assistance.

closed 2018-02-07 by eerimoq