cantools/cantools #22: Invalid DBC syntax: 'BA_REL_ "GenSigComment" BU_SG_REL_ ICCM_CE SG_ 569 SomeString >!<"";': Expected positive integer."

sietschie, 2018-02-07:
Using Python 3.4.4 and cantools 16.2.0 I get the following error when trying to load my DBC:

> cantools.db.UnsupportedDatabaseFormatError: DBC: "Invalid DBC syntax at line 5569, column 80: 'BA_REL_ "GenSigComment" BU_SG_REL_ ICCM_CE SG_ 569  SomeString >!<"";': Expected positive integer."

Is it possible to add support for this?

eerimoq, 2018-02-07:
Almost anything is possible. Can you provide a DBC-file I can use for testing, to make sure the implementaiton works?

sietschie, 2018-02-07:
I am afraid I am not allowed to give out the original DBC file. And I don't know enough about DBCs to produce my own file. Maybe you can point me in the right directions so I can make the changes myself?

eerimoq, 2018-02-07:
Have a look here: https://github.com/eerimoq/cantools/blob/master/cantools/db/formats/dbc.py#L305-L313

There is the part of the grammar where your DBC fails. Looks like the grammar should not only allow positive integer, but maybe floating point numbers?

eerimoq, 2018-02-07:
I suggest that you add the failing DBC-line here: https://github.com/eerimoq/cantools/blob/master/tests/files/foobar.dbc#L83

And then modify the tests in tests/test_cantools.py to make them pass.

closed 2018-02-07 by eerimoq

DC-Melo, 2020-12-03:
my database is like this：
 BA_DEF_ BU_  "NmStationAddress" HEX 0 0;
   1 BA_DEF_ BU_  "NodeLayerModules" STRING ;
1811 BA_DEF_REL_ BU_BO_REL_  "GenMsgCycleTimeFast" INT 0 65535;
   1 BA_DEF_REL_ BU_BO_REL_  "GenMsgNrOfRepetition" INT 0 65535;
   2 BA_DEF_REL_ BU_BO_REL_  "GenMsgDelayTime" INT 0 65535;
   3 BA_DEF_REL_ BU_BO_REL_  "GenMsgStartDelayTime" INT 0 65535;
   4 BA_DEF_DEF_  "BusType" "";
   5 BA_DEF_DEF_REL_ "GenMsgCycleTimeFast" 0;

so：i change the code to：
        attribute_definition_rel = Sequence(
            'BA_DEF_REL_',
            Optional(choice('BU_SG_REL_','BU_BO_REL_')),
            'STRING',
            'WORD',
            choice(DelimitedList('STRING'), OneOrMore('NUMBER')),
            ';')

now：it is ok

Guemi, 2021-08-24:
I've modify it the dbc.py file to the following:
`        attribute_definition_rel = Sequence(
            'BA_DEF_REL_',
            Optional(choice('BU_SG_REL_', 'BU_BO_REL_')),
            'STRING',
            'WORD',
            Optional(choice(DelimitedList('STRING'), OneOrMore('NUMBER'))),
            ';')`
to support a dbc that looks like this:
`BA_DEF_REL_ BU_BO_REL_  "Frame1" STRING ;
BA_DEF_REL_ BU_BO_REL_  "Frame2" HEX 0 1407;
BA_DEF_REL_ BU_BO_REL_  "Event1" INT 10 10000;
BA_DEF_REL_ BU_SG_REL_  "Signal1" ENUM  "Yes","No";`

PR #23 (2018-02-07): added grammar for SIG_GROUP_ and optional quoted string at the end of…
This commit fixes #22 
There was another parsing error: the SIG_GROUP_ keyword was not recognized. I added grammar for that to the best of my abilities and now at least the parsing passes without error.