cantools/cantools #24: For database objects, attribute_definition_defaults and attribute_definitions should defaults to dicts, not lists

rjw245, 2018-02-09:
Otherwise, I see the following error:
    default_send_type = database.attribute_definition_defaults['GenMsgSendType']
TypeError: list indices must be integers, not str

rjw245, 2018-02-09:
This also applies to File objects

eerimoq, 2018-02-09:
These attributes in the Database class is an ugly hack to preserve the information from a parsed DBC-file. The interface to access such attributes should be reworked to be clean and useful. I have not made up my mind yet of how to design the final interface. Any ideas?

rjw245, 2018-02-12:
I think in either case, you're mapping the attribute name to a value right? Dictionary seems most appropriate for both, IMO.

eerimoq, 2018-02-12:
Does it help you if they are both dictionaries? What is the use case? As said, the storage of attributes should be redesigned at some point, and maybe dictionaries will be used then.

Will you submit a PR with a fix for the wrong default values?

closed 2018-02-15 by eerimoq