cantools/cantools #27: Attribute definitions do not appear in output of Database::as_dbc_string()

rjw245, 2018-02-25:
I am instantiating a Database with an attribute definition like so:

```
Database(messages, nodes, buses, version,
                           attribute_definitions=[
                                                      ('BA_DEF_',
                                                       'BO_',
                                                       'GenMsgCycleTime',
                                                       'INT',
                                                       [ '0',
                                                         '1000000'
                                                       ],
                                                      )
                                                 ]
)
```

And one of the messages has a non-null cycle_time attribute set, which does show in the DBC. However, the attribute definition itself does not. I can't seem to figure out why.

I did notice that the _dump_attribute_definitions() function in formats/dbc.py seems to expect attribute definitions in a different format than they show when going FROM  a dbc TO  a Database object; see below:

```
>>> db = cantools.db.load_file("mydbc.dbc")
>>> print db.__dict__
...
 '_attribute_definitions': [(['BA_DEF_', 'BO_', 'GenMsgCycleTime', 'INT', (['0', '1000000'], {})], {})],
...
```
As you can see there is a lot more nesting going on than what I passed to the constructor -- the simpler format seems to be what _dump_attribute_definitions expects.

rjw245, 2018-02-25:
Never mind, I think I had missed something in my testing -- instantiating the Database's attribute_definitions the way I did DOES cause them to appear in the DBC. However, they did not when I passed them in in the format they appear when loading FROM a DBC. I think the two different formats should be reconciled in some way.

eerimoq, 2018-02-25:
Indeed, the attribute storage is a hack. It is not recommended to be used before redesigned. A redesign is not likely to happen anytime soon unless someone submits a pull request with the implementation.

closed 2018-03-01 by eerimoq