cantools/cantools #30: 'utf-8' codec can't decode byte

travlytle, 2018-04-26:
When trying to use:
candump can0 | cantools decode --single-line ../j1939.dbc

I get the following error:
'utf-8' codec can't decode byte 0xb0 in position 4832: invalid start byte

eerimoq, 2018-04-26:
I implemented what I think is a fix for your error. Install version 20.3.1 with `pip3 install -U cantools`.

travlytle, 2018-04-27:
Still dumps to console

pi@raspberrypi:~ $ cdump can0 | cantools decode --single-line j1939.dbc
'utf-8' codec can't decode byte 0xb0 in position 4832: invalid start byte
pi@raspberrypi:~ $

travlytle, 2018-04-27:
Actually looks like it didn't upgrade, I'll try again tomorrow.

cantools --version still shows 20.3.1

eerimoq, 2018-04-27:
Ok.

As a workaround you can remove all degree-symbols in the DBC-file. That said, lets continue the troubleshooting to solve the issue.

I can successfully read [a DBC-file](https://github.com/eerimoq/cantools/blob/master/tests/files/vehicle.dbc) with a degree-symbol on my computer.

travlytle, 2018-04-27:
So I got cantools to upgrade and the error about utf-8 went away.  But now I don't see any decoding.

pi@raspberrypi:~ $ candump can0
  can0  1CF004FF   [8]  00 00 00 00 00 00 00 00
  can0  1CF004FF   [8]  00 00 00 00 00 00 00 00
  can0  1CF004FF   [8]  00 00 00 00 00 00 00 00
  can0  1CF004FF   [8]  00 00 00 00 00 00 00 00
  can0  1CF004FF   [8]  00 00 00 00 00 00 00 00
  can0  1CF004FF   [8]  00 00 00 00 00 00 00 00
  can0  1CF004FF   [8]  00 00 00 00 00 00 00 00
^C
pi@raspberrypi:~ $ candump can0 | cantools decode --single-line j1939.dbc
^C
pi@raspberrypi:~ $ cantools --version
20.3.1
pi@raspberrypi:~ $

travlytle, 2018-04-27:
I let the following command set awhile:
candump can0 | cantools decode --single-line j1939.dbc

And I get this:
pi@raspberrypi:~ $ candump can0 | cantools decode --single-line j1939.dbc
DBC: "Invalid DBC syntax at line 8870, column 80: 'CM_ SG_ 2633785086 StOfHealth "This signal represents the expected remaining \">!<Life\" of the High Voltage Energy Storage 1 pack (or system), in percent.  This value should begin at 100%, and end at 0% when �no� energy is able to be put in or taken out.  �No� could be absolute zero, ...";': Expected ";".", KCD: "syntax error: line 1, column 0", SYM: "Only SYM version 6.0 is supported."

travlytle, 2018-04-27:
So I tried removing the degree-symbols and I still got this error:
pi@raspberrypi:~ $ candump can0 | cantools decode --single-line j1939_test.dbc
DBC: "Invalid DBC syntax at line 8870, column 80: 'CM_ SG_ 2633785086 StOfHealth "This signal represents the expected remaining \">!<Life\" of the High Voltage Energy Storage 1 pack (or system), in percent.  This value should begin at 100%, and end at 0% when �no� energy is able to be put in or taken out.  �No� could be absolute zero, ...";': Expected ";".", KCD: "syntax error: line 1, column 0", SYM: "Only SYM version 6.0 is supported."

So I removed all the \" and I got the following:
pi@raspberrypi:~ $ candump can0 | cantools decode --single-line j1939_test.dbc
invalid literal for int() with base 10: '0M'

eerimoq, 2018-04-27:
I'll have a look at it soon :)

travlytle, 2018-04-27:
It's now working enough I can use it.  It isn't decoding my messages because it seems the CANID has to be an exact match which is OK.  Beyond that you would be getting into J1939 standard specifics.

Currently this is what I'm seeing:
 can0  1CF004FE   [8]  00 00 00 00 00 00 00 00 :: Unknown frame id 485491966
 
It's not decoding because the frame id (CANID) isn't an exact match as stated.  You could decode based on PGN, but that will not cover all the specific j1939 frames.  You would have to dig into the J1939 standard to cover all/most things.  You may look at the https://rbei-etas.github.io/busmaster/ project as it supports J1939 some what.

Anyway it's working enough now I can use it, anything else would just be added features.

I realized it's slow because of the parsing, it's a huge dbc but has almost everything in it.  Would be great if it's faster but not necessary.  I guess it would be best to data log the device in question then make just a dbc for that device.  (which is what I will end up doing).

Thank you for the quick fix/update!

eerimoq, 2018-04-27:
What about adding an optional argument to `load_file()`, `frame_id_mask`, which can be used to ignore certain bits in the frame id when decoding (and possibly other places)? Does it make any sense to add that?

eerimoq, 2018-04-28:
I added a frame id mask argument that can be used to ignore bits in the frame id. I'm not sure it will solve the general case, but at least it works for the example you gave.

It's part of version 20.4.0.

```
$ echo " can0  1CF004FE   [8] 00 00 00 00 00 00 00 00" | cantools decode --frame-id-mask 0x03ffff00 j1939.dbc
 can0  1CF004FE   [8] 00 00 00 00 00 00 00 00 ::
EEC1(
    EngTorqueMode: 0,
    ActlEngPrcntTorqueHighResolution: '+0.000%' %,
    DriversDemandEngPercentTorque: -125 %,
    ActualEngPercentTorque: -125 %,
    EngSpeed: 0.0 rpm,
    SrcAddrssOfCntrllngDvcForEngCtrl: 0,
    EngStarterMode: 'start not requested',
    EngDemandPercentTorque: -125 %
)
```

closed 2018-04-29 by eerimoq