cantools/cantools #31: Arxml support

danielhrisca, 2018-05-02:
Hello Erik,

what do you think about adding arxml format to the supported database formats?

https://automotive.wiki/index.php/AUTOSAR_XML_Schema

danielhrisca, 2018-05-03:
I could not find the specification for this format but a starting point could be _canmatrix_ 

https://github.com/ebroecker/canmatrix

eerimoq, 2018-05-03:
Hmms... maybe you can use canmatrix instead? =)

danielhrisca, 2018-05-04:
I'm already using _canmatrix_ because it supports arxml. Initially I've used _cantools_ to handle DBC files.

I appreciate _cantools_ and I think it will be a very nice feature to have.

eerimoq, 2018-05-04:
I might implement it. Though, it would be very helpful if you could start by writing a bunch of tests to verify that the implementation works as expected. It's unlikely I'll start the implementation without provided tests as I have no experience of ARXML and no possibility to easily test it.

eerimoq, 2018-05-08:
Any progress?

closed 2018-05-11 by eerimoq