cantools/cantools #34: as_dbc_string()  database's method loses the extended flag

dmyger, 2018-05-16:
Sample of code with the issue:
```
db = cantools.db.Database()
db.add_dbc_file('some.dbc')
ext_msg = db.get_message_by_name('SomeExtendedMessage')

if ext_msg.is_extended_frame:
    db_out = cantools.db.Database()
    db_out.add_message(ext_msg)
    with open('result.dbc', 'w') as dbc_result:
        dbc_result.write(db_out.as_dbc_string())
```
Check `frame_id` in the result output and see that `0x80000000` bit is absent.
And if open `result.dbc` with your library it also show the `is_extended_frame` flag is `False`:
```
db = cantools.db.Database()
db.add_dbc_file('result.dbc')
for m in db.messages:
	print m
```

eerimoq, 2018-05-16:
The `as_`-methods are not the primary feature of this package, and should probably be removed since they are not fully functional. Either that, or I should add a limitation in the README notifying the user about their flaws. That said, feel free to implement the missing feature if you find it useful.

dmyger, 2018-05-16:
It seems, that exactly for that issue, it would be easy set a high bit for the extended frame in file `cantools/db/formats/dbc.py` for each case where is used `message.frame_id`, something like this
```
    ...
    msg.append(fmt.format(frame_id=message.frame_id | 1<<31 if message.is_extended_frame else 0,
    ...
```
Instead of removing the working functionality.

eerimoq, 2018-05-16:
Feel free to create a pull request with the change and I'll review and later merge it. Make sure to modify/add tests to ensure the issue is not reintroduced in the future. Thanks!

eerimoq, 2018-05-19:
Part of release 20.9.0.

closed 2018-05-19 by eerimoq