cantools/cantools #42: Big Endian Decoding Errors

maxchill, 2018-06-16:
I believe I've identified a few big endian related decoding errors or improvements.

1. Skip bigendian signal start calculation / decode logic if signal <= 8 bits as "Endianness refers to the sequential order in which bytes are arranged into larger numerical values "
2. When the signal at the end of a message is bigendian a sanity check fails 
   "unpack requires at least 71 bits to unpack (got 64)".  I included a unit test triggering this, but have not decided on the correct fix yet, so wanted to open the topic for discussion to get the opinion of the maintainer.

I've started a branch here, and can submit a PR after I get your opinion on fixing issue #2.
https://github.com/maxchill/cantools/tree/bugfix/bigendian_decoding_errors

juleq, 2018-06-16:
Hi. I have not used KCD but the last time that I had to cope with byte order and bit numbering with the DBC format, I found this writeup particulary useful:
- [Signal Byte Order](https://github.com/ebroecker/canmatrix/wiki/signal-Byteorder)

They also sum up which format takes which approach. Maybe that helps.

eerimoq, 2018-06-17:
You can also have a look here: http://cantools.readthedocs.io/en/latest/#cantools.db.Signal

Signal placement in your ``test_big_endian_signal_two_byte_decode()`` test is as below.

The big endian signal placement is very counter intuitive, but I have no plans on changing it. To place a big endian signal at the end, the start bit should be 55, not 48.

Start bit 48:

```
Byte:       0        1        2        3        4        5        6        7
       +--------+--------+--------+--------+--------+--------+--------+--------+
       |        |        |        |        |        |        |       x|xxxxxxxx|xxxxxxx
       +--------+--------+--------+--------+--------+--------+--------+--------+
Bit:    7      0 15     8 23    16 31    24 39    32 47    40 55    48 63    56
```

Start bit 55:

```
Byte:       0        1        2        3        4        5        6        7
       +--------+--------+--------+--------+--------+--------+--------+--------+
       |        |        |        |        |        |        |xxxxxxxx|xxxxxxxx|
       +--------+--------+--------+--------+--------+--------+--------+--------+
Bit:    7      0 15     8 23    16 31    24 39    32 47    40 55    48 63    56 
```

eerimoq, 2018-06-17:
I implemented a fix where the KCD big endian signal offset is converted the the cantools start bit definition. That is, KCD offset 0 becomes start bit 7. The Kayak project, that originally defined the KCD format, implements signal unpacking as: https://github.com/dschanoeh/Kayak/blob/5c45f9249d0c4cb5ee04f28a3ff14a1400858213/Kayak-core/src/main/java/com/github/kayak/core/description/SignalDescription.java#L279-L301

I think the new cantools implementation does the same thing. Feel free to try it out and report any issues.

See c30a631aed40a10945dca075f435ae983ed6e99c for the fix.

maxchill, 2018-06-17:
Thanks a lot for sharing links to help me understand the correct decoding behavior.  KCD unlike DBC, isn't as established so it's hard to determine what the correct behavior should be.

The following info from [Canmatrix Signal ByteOrder](https://github.com/ebroecker/canmatrix/wiki/signal-Byteorder) was the key revelation for me:
> **With "consistent" format/numbering, bit numbering follows the byte format. MSB0 for big endian, LSB0 for little endian.**
> 
> KCD, SYM (consistent bit numbering)
>     little,LSB0
>     big,MSB0
> 
> DBC, ARXML (OSEK bit numbering)
>     little,LSB0
>     big,LSB0 (inconsistent)

However, the KCD XML schema seems to make no mention of bit order consistency.

https://github.com/dschanoeh/Kayak/blob/master/Kayak-kcd/src/main/resources/com/github/kayak/canio/kcd/loader/Definition.xsd

It seems to me the offset section of the KCD schema should mention if endianess is big, the offset is based on MSB0
```
<xs:attribute name="offset" use="required">
      <xs:annotation>
        <xs:documentation>Least significant bit offset of the signal relative to the least significant bit of the messages data payload.</xs:documentation>
</xs:annotation>
```

@eerimoq If it is infact the case that KCD has consistent bit numbering, the commit you made makes sense / works for me.

eerimoq, 2018-06-17:
@maxchill Please open an issue in the [KCD project](https://github.com/julietkilo/kcd) ask for clarification of the signal endianess and placement. It would be nice if the cantools implementation is correct according to the KCD format specification.

eerimoq, 2018-06-18:
Otherwise, please close the issue and we'll deal with this in the future.

eerimoq, 2018-06-18:
@maxchill Otherwise, please close the issue and we'll deal with this in the future.

eerimoq, 2018-06-18:
The fix implemented as part of this issue is part of release 22.0.0 on PyPI.

maxchill, 2018-06-19:
I will follow through / get to the bottom of this later this week.  Thanks.

eerimoq, 2018-06-24:
I will close this if there is no feedback soon.

closed 2018-06-26 by eerimoq