cantools/cantools #44: Can't load DBC file with extended frames

dmyger, 2018-06-20:
After updating up to 22 version, it was lose ability to load DBC files with extended frames.
```Python
import cantools
db = cantools.db.load_file(some_extended_db)
```
If frame ID has high bit is on, I get assert error on such frame:
```Python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "build/bdist.linux-x86_64/egg/cantools/db/__init__.py", line 67, in load_file
  File "build/bdist.linux-x86_64/egg/cantools/db/__init__.py", line 89, in load
  File "build/bdist.linux-x86_64/egg/cantools/db/__init__.py", line 123, in load_string
  File "build/bdist.linux-x86_64/egg/cantools/db/database.py", line 123, in add_dbc_string
  File "build/bdist.linux-x86_64/egg/cantools/db/formats/dbc.py", line 1139, in load_string
  File "build/bdist.linux-x86_64/egg/cantools/db/formats/dbc.py", line 702, in _load_attributes
KeyError: 3221225472
```
But, if I remove only high bit from frame id, the same dbc successfully loaded.

eerimoq, 2018-06-20:
@juleq Can you have a look at this?

juleq, 2018-06-20:
I am on it. So far not able to reproduce. @dmyger Could you provide a minimal dbc failing?

juleq, 2018-06-20:
This is what I did for reproduction so far:
- Changed TheMessage in tests/files/attributes.dbc to extended
- Changed all attributes and their definitions accordingly
All tests complete.

juleq, 2018-06-20:
Since you are on linux, I assume that you do not have CANdb++ to open for a test at hand?

I am able to reproduce by doing the following:
- Change one attribute difinition to a non existent frame id
- CANdb++ will not open the file anymore, reporting the offending frame id
- cantools will behave similar by reporting the key error

This seems to exactly reproduce hat you are seeing. If you can confirm that assumption, we could think of a more meaningful error message.

eerimoq, 2018-06-20:
Reading [the attribute load code](https://github.com/eerimoq/cantools/blob/master/cantools/db/formats/dbc.py#L679-L713) I cannot understand why it works with extended id, but fails with standard id. Though, it may fail because the signal attributes are before the message attributes in your DBC-file. In that case the message attribute the signal needs is not found, giving the error as in your backtrace.

juleq, 2018-06-20:
He said it would fail with extended.

You are right in that this code is sensitive to sequence: Definitions have to be before the attributes. That is also true for CANdb++. If it encounters an attribute it has no definition for, it will error out.

eerimoq, 2018-06-20:
Swapped extended and standard, sorry. Edited the comment to make it correct.

juleq, 2018-06-21:
@dmyger Could you give us feedback on this? Thank you for your report.

eerimoq, 2018-06-24:
Will close this if there is no feedback soon.

JulianWgs, 2018-06-26:
I think Im having the same issue:

You can download a test dbc here: 

https://www.formulastudent.de/fileadmin/user_upload/all/2017/important_docs/FSG2017_Data_Logger_EM_data_V1.0.dbc

Test code is this:
```
import cantools
db = cantools.db.Database(messages=None, nodes=None, buses=None, version=None)
db.add_dbc_file("FSG2017_Data_Logger_EM_data_V1.0.dbc")
```

This is the error message:

```
Traceback (most recent call last):
  File "C:\XXX\XXX\XXX\XXX\test.py", line 5, in <module>
    db.add_dbc_file("FSG2017_Data_Logger_EM_data_V1.0.dbc")
  File "C:\Users\XXXX\AppData\Local\Programs\Python\Python36\lib\site-packages\cantools\db\database.py", line 111, in add_dbc_file
    self.add_dbc(fin)
  File "C:\Users\XXXX\AppData\Local\Programs\Python\Python36\lib\site-packages\cantools\db\database.py", line 97, in add_dbc
    self.add_dbc_string(fp.read())
  File "C:\Users\XXXX\AppData\Local\Programs\Python\Python36\lib\site-packages\cantools\db\database.py", line 123, in add_dbc_string
    database = dbc.load_string(string)
  File "C:\Users\XXXX\AppData\Local\Programs\Python\Python36\lib\site-packages\cantools\db\formats\dbc.py", line 1139, in load_string
    attributes = _load_attributes(tokens, attribute_definitions)
  File "C:\Users\XXXX\AppData\Local\Programs\Python\Python36\lib\site-packages\cantools\db\formats\dbc.py", line 702, in _load_attributes
    if 'signal' not in attributes[frame_id]:
KeyError: 3221225472

Process returned 1 (0x1)        execution time : 0.476 s
```

Python 3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 17:00:18) [MSC v.1900 64 bit (AMD64)] on win32

Also I would suggest adding tests for this.

I hope this helps :)

juleq, 2018-06-26:
Hello @JulianWgs. Thanks for reporting. This bug is waiting for merge (#45) already. It happend due to a missing initialisation when the load_attributes method hit a signal attribute without having a message attribute with the same id before.

If you need a quick fix, just look at PR #45, commit 9b37564.

juleq, 2018-06-26:
@JulianWgs Concerning tests: The attributes test covers this case when PR is merged, I have added such a signal attribute.

JulianWgs, 2018-06-26:
I just moved back to 21.0.3 and will update to the latest release as soon as this is fixed. Thanks for caring and the quick reply! Its also great to hear that you improved tests.

eerimoq, 2018-06-26:
Fixed in cantools version 22.0.1, available on PyPI.

closed 2018-06-26 by eerimoq

smpshehan, 2021-03-26:
I used latest cantools python package v36.2 to decode canbus message. I used a dbc file created by me for testing. I copied a sample file. When I use can id like 419358976, I get error. But for smaller can ids like 350, it works. Does cantools fail for extended frame ids? how do I get this working?

my code which is failing for extended id's is as follows:

```
db = cantools.database.load_file('.\\src\\test\\resources\\j1939.dbc')
print(db.decode_message(419358976,b'\xff\xff\xff\xc0\x0c\xff\xff\xff'))
```

error: `cantools.database.errors.Error: Standard frame id 0x18fee900 is more than 11 bits in message EEC1.`

pmoriano, 2021-05-11:
@smpshehan thanks for sharing. Did you get that working? Facing a similar issue.

pmoriano, 2021-05-11:
Sorry @juleq. I updated my previous comment. Thanks again.

juleq, 2021-05-11:
You could post the id that you are using or the command line you are trying to execute. Similar to the information @smpshehan had provided in his post.

juleq, 2021-05-11:
Ah, sorry. You likely do not have his problem. Without seeing your error message I would guess that probably your dbc is hand edited and probably has attributes or their definitions in wrong order? Do you have CANdb++ at hand to open the dbc file? If it can open it without erroring out, I would expect cantools to be able to, too.

You can also post your dbc (I assume you are using a dbc?).

juleq, 2021-05-11:
Background: KeyError here means that you have passed in an id that is not contained in the message dictionary that cantools did set up for you when loading the dbc. 

If you do not want to post the whole dbc, I would recommend posting the message and signal definition for that id from the dbc. It starts with BO_ and then your id, the signals following with SG_.

Further, if cantools outputs some additional context for the error, you could post that, too.

pmoriano, 2021-05-17:
Thanks for your help @juleq. 

Please see below more details about the error (a subsection of the BDC):

DBC:

BO_ 2297441790 ID1: 7 Vector__XXX
 SG_ S1 : 54|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ S2 : 52|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ S3 : 50|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ S4 : 48|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ S5 : 32|16@1+ (0.01,-320) [-320|322.55] "m/s≤" Vector__XXX
 SG_ S6 : 0|16@1+ (0.01,-320) [-320|322.55] "m/s≤" Vector__XXX
 SG_ S7 : 16|16@1+ (0.01,-320) [-320|322.55] "m/s≤" Vector__XXX

The code I am using and subsequent error is below.

Code:

```
import can
import cantools

j1939db = cantools.database.load_file("path_to_dbc") 

aid = 2297441790
msg = can.Message(arbitration_id=aid, data=[173, 255, 173, 0, 14, 1, 1, 100])

j1939db.decode_message(aid, msg.data)
```

Error:

`KeyError: 2297441790`

Note that the ID is 32-bit long. Thank you again.

juleq, 2021-05-17:
Hi.

Changing aid = 2297441790 to aid = 149958142 will make it work.

Background: The dbc adds 0x80000000 to the id to indicate it is extended. You do not need that when decoding because cantools already has the definition and thus knows that this message has an extended id and will treat it accordingly.

BR

juleq, 2021-05-17:
BTW: If you enter the id as hex literal, it is more recognizable: aid=0x08F02DFE...

pmoriano, 2021-05-17:
Thanks @juleq. It makes sense. Now it works. I hope this helps others.