cantools/cantools #46: Include VAL_TABLE_ in the dbc dump.

juleq, 2018-06-27:
Value tables are currently omitted when a db is dumped to dbc format. Include them. Further a proper access interface to these value tables (e.g. in the dbc specifics) might make sense.

juleq, 2018-06-27:
Since I do not have sufficient permissions, I cannot label this as enhancement.

karlding, 2018-11-06:
Would something like this work? @eerimoq I can work on this and submit a PR if you're okay with formatting it like so:

```
================================= Messages =================================

  ------------------------------------------------------------------------

  Name:       DRIVER_HEARTBEAT
  Id:         0x64
  Length:     1 bytes
  Cycle time: 1000 ms
  Senders:    DRIVER
  Layout:

                          Bit

             7   6   5   4   3   2   1   0
           +---+---+---+---+---+---+---+---+
     B   0 |<-----------------------------x|
     y     +---+---+---+---+---+---+---+---+
     t       +-- DRIVER_HEARTBEAT_cmd
     e

  Signal tree:

    -- {root}
       +-- DRIVER_HEARTBEAT_cmd

  Signal choices:

    DRIVER_HEARTBEAT_cmd
      2 DRIVER_HEARTBEAT_cmd_REBOOT
      1 DRIVER_HEARTBEAT_cmd_SYNC
      0 DRIVER_HEARTBEAT_cmd_NOOP

  ------------------------------------------------------------------------
```

Or we can talk about other ways of formatting this if you don't like that!

eerimoq, 2018-11-06:
Looks pretty good! Though, I think I prefer the choices in ascending numeric order and four space indentation.

```
    DRIVER_HEARTBEAT_cmd
        0 DRIVER_HEARTBEAT_cmd_NOOP
        1 DRIVER_HEARTBEAT_cmd_SYNC
        2 DRIVER_HEARTBEAT_cmd_REBOOT
```

eerimoq, 2018-11-13:
`VAL_TABLE_` part of version 28.16.0.

closed 2018-11-13 by eerimoq