cantools/cantools #47: Unnecessary type conversion

BayarjavkhlanBatbaatar, 2018-06-28:
When there is a default string attribute which is numeric, it is converting it to int. As result there is type conflict.

Let's say there is a .dbc file with attribute with type of string and its default value is "100". When you load the .dbc file and then do db.as_dbc_string(), it would not have default value for the attribute. 

https://github.com/eerimoq/cantools/blob/d1127b07efea9ec74844afc9d36f0f7095160d86/cantools/db/formats/dbc.py#L522

juleq, 2018-06-28:
That is correct. I will fix that soon. Thank you for reporting.

BayarjavkhlanBatbaatar, 2018-06-29:
Thank you so much. You guys are great.

closed 2018-06-29 by BayarjavkhlanBatbaatar

eerimoq, 2018-06-29:
The fix is part of release 22.0.2, available on PyPI!