cantools/cantools #48: Proposal to integrate CDD and ODX

anxuae, 2018-06-28:
Hi,

I'm working on CAN projects and I'm currently using the `cantools` lib to parse my `DBC` files and it's work great! (I'm using it as a complement to `python-can` lib ) .

I'm wondering if you plan to integrate parsers for `CDD` (CANdelaStudio diagnostic description) files and maybe a more open format like `ODX` (Open diagnostic data exchange) also?

I could give you some help about the `CDD` file format.

ax

eerimoq, 2018-06-28:
Hello,

there are no such plans, but if you are willing to invest time into implementing it I can gladly provide guidance, if so needed.

Alternatively, if it's not too much work, I can implement it. Any reference data format description would be much appreciated to make sure the implementation is correct.

eerimoq, 2018-06-30:
@anxuae Any updates?

anxuae, 2018-06-30:
Hi @eerimoq,

Here if someone who did the trick in labview: https://forums.ni.com/t5/Community-Documents/Use-inbuilt-LabVIEW-XML-functions-to-read-Vector-CANdela-cdd/ta-p/3511391

The CDD format is just XML, thus `xml.etree.ElementTree` is OK to parse the file. The complexity is to understand the tokens as CDD is a Vector format and thus not really open format (like DBC in fact)

A other point is that a new `Database` class will be required as this file describes the diagnostic `services` available in the car. This database shall permit to encode/decode diagnostic requests/answers from/to one/several ECU.

I will attach a file as example

anxuae

eerimoq, 2018-06-30:
I started to implement support for CDD and pushed the first commit, e06f7555c26fbb4c81a792d05086d5aac617effa. For simplicity in this early design phase CDD uses that same database as already in place. Each `DIAGINST` becomes a message, and its members becomes signals. If there is a `UNION` only the `STRUCT` members are used.

There is no support for anything but encoding and decoding of the service data. See the [test suite](https://github.com/eerimoq/cantools/blob/master/tests/test_database.py#L1944-L2004) for example usage.

Is there anything that has to be added or changed to make the library useful?

Would be nice if you can write an example showing how you want to use the library.

eerimoq, 2018-07-01:
@anxuae I'm waiting for your feedback before continuing.

anxuae, 2018-07-01:
Ok, I will try to send you an example soon.
I'm currently doing some tests.

eerimoq, 2018-07-03:
@anxuae Create a pull request with the example script once ready.

eerimoq, 2018-07-04:
Any progress?

eerimoq, 2018-07-05:
@anxuae Anything?

closed 2018-07-06 by eerimoq

fuchsde, 2021-01-18:
@eerimoq @anxuae Thanks for pushing the parsing of cdd files. I had a look into the source code. I would be eager to read out the services provided with each DID as well, but I do not have an idea how to translate the vector format into the CAN request, negative and postive response. Can someone help me out?

eerimoq, 2021-01-18:
I don't have time to help, sorry.

andlaus, 2021-03-09:
Please *don't* implement isotp within cantools, as this is out of scope. Instead, use an external module like [can-isotp](https://pypi.org/project/can-isotp/) for the interaction with the bus (pretty much like cantools provides the interpretation for can signals while the actual bus interfacing is done by an external module e.g., `python-can`)

andlaus, 2021-03-09:
(also note that starting the more robust kernel module for isotp has been merged into the 5.10 mainline kernel and that the 'can-isotp' module seems to have transparent support for it.)