cantools/cantools #50: Manage Vector__XXX receivers in DBC

anxuae, 2018-07-01:
In DBC files when a signal has one receiver which `Vector__XXX`, that means this signal is received by all nodes.

Is it possible that `signal.receivers` return the complete list of node names present on the same bus instead of `Vector__XXX`?

ax

eerimoq, 2018-07-01:
Does it apply to `message.senders` as well?

eerimoq, 2018-07-01:
Are you sure that `Vector__XXX` means all nodes, and not that the receiver nodes are unspecified?

anxuae, 2018-07-01:
Ok after checking different sources, I'm not sure any more.
The majority says that it means undefined, thus I think that you can close this issue.
Sorry about that.

eerimoq, 2018-07-01:
No problem. My guess is that Vector__XXX is a dummy placeholder to make the DBC-format easier to parse.

closed 2018-07-01 by eerimoq