cantools/cantools #52: Approaches for handling multiple concurrent databases

pevsonic, 2018-07-18:
Heya, 

Great project! I've been using it to support dbc loading for an ncurses based can monitoring tool I've written. However we have a situation where we have multiple dbc files for different components within the network as a whole (yes there are inevitably conflicts...!). This means that when monitoring the network I have to load multiple dbc files. Has this use case been given any consideration as yet?

There are two possible approaches I see are : 

1) Merging dbcs ; I was originally instantiating one database object and was just calling add_dbc_file() multiple times. This used to appear to work in the past (on the surface at least!) but has now been hobbled by recent changes to database.py where refresh clears _frame_id_to_message every time add_dbc_string() is called. (This can be bodged by changing add_xxx_string() to effectively merge in new databases from load_string rather than re-creating). I quite like this approach as it highlights clashes between multiple dbcs and keeps it convenient. However there are lots of potential pitfalls and loading all of our databases takes nearly 40 secs on startup of the tool :-O 

2) Load databases independently ; so as a second workaround I wrote a small wrapper class that creates a new database every time I load in a dbc file and then provides an abstraction of interfaces such as get_message_by_frame_id() that iterates through all loaded databases. The downside is that if a frame id (or other object) exists in multiple databases, there's no clash detection ... However parsing is much faster : the same set of dbcs load in 15 secs instead of 40...

Do you have any thoughts?

eerimoq, 2018-07-18:
Hello,

I'm glad you like the project!

As for 1. above, I'll do my best reimplementing support for adding multiple files to the same database object. Adding all messages should be easy. It's a bit complicated to append nodes, version and DBC attributes. Does it solve your use case if I start by only implementing adding messages from multiple DBC-files? I guess it does because that's what was previously implemented.

`cantools` is using `pyparsing` to parse DBC-files, and it is often very slow. It will be even slower in the latest release because it now supports `//` comments in the DBC files.

There is another parsing library called [Lark ](https://github.com/lark-parser/lark)that I'm currently evaluating. Its LALR parser should be **a lot faster** that the current parser. If the performance is as big as expected the new parser should be used by `cantools` in a few days.

eerimoq, 2018-07-18:
1. is fixed in release 28.0.1, available on PyPI.

eerimoq, 2018-07-19:
Seems Lark is pretty slow as well, at least for a Lark novice like me.

Using regular expressions instead of a parser library seems to be the only viable option to create a fast parser.

fragmuffin, 2018-07-19:
I started to write a regex-based parser in https://github.com/fragmuffin/dbcparser
But while researching, I found this library and abandoned development.
(see [`Signal` class](https://github.com/fragmuffin/dbcparser/blob/master/src/dbcparser/parser.py#L386) for an example)

I'd be happy to put that work into this project if you're interested.
I haven't stress tested it... what's your benchmark for speed?

eerimoq, 2018-07-20:
I'm not sure yet how I want to proceed. I like expressing the DBC grammar with pyparsing, but the user does not care about that, she wants a fast parser. Give me a few days to think about this.

I use a DBC-file I cannot share for my benchmark. It's a few thousand lines, and takes about a second to parse.

fragmuffin, 2018-07-20:
@eerimoq 

> I'm not sure yet how I want to proceed. I like expressing the DBC grammar with pyparsing

I understand, I haven't tried  `pyparsing` yet, I should definitely give it a go.

> I use a DBC-file I cannot share for my benchmark. It's a few thousand lines, and takes about a second to parse.

You can try it out on your machine with.

```bash
git clone git@github.com:fragmuffin/dbcparser.git
cd dbcparser/src
python
```

```python
>>> filename = '/your/dbc/file.dbc'  # mine's only a few hundred lines long

>>> from dbcparser.parser import DBCParser
>>> parser = DBCParser(open(filename, 'r'))
>>> (start_time, dbc, end_time) = (time.time(), parser.parse(), time.time())
>>> end_time - start_time
0.046880483627319336
```

That does all the parsing, but I never finished the container classes.
So the values are all parsed, but they're just left to be garbage collected when the parser's finished.

The text parsing is most of the work, so the timing result is still very much relevant.

Anyway, I'm not pushing, just letting you know that the regular expressions are all done, working, and thoroughly tested if you choose to go that way.

eerimoq, 2018-07-21:
@pevsonic 

There is a new DBC parser implemented on the re-branch, found here: https://github.com/eerimoq/cantools/tree/re

Can you compare its performance to version 28.1.0 and post the results here?

eerimoq, 2018-07-24:
Faster DBC loading is part of version 28.2.0. It's roughly three times faster than before.

closed 2018-07-24 by eerimoq

pevsonic, 2018-09-03:
@eerimoq - sorry for delay, I've been away avoiding computers for a month! 

So, I've updated my copy and am now at 28.7.2. My project loads 27 different databases which at the time of originally writing loaded in 39 seconds ; this is now 24 seconds so I must assume that there's an additional bottle-neck somewhere. 

I did wonder if maybe using multiprocessing with database.add_dbc_file() would be an option here but I suspect that may be asking for trouble?

eerimoq, 2018-09-03:
Hello,

does this work for you:

``` python
import cantools
from multiprocessing import Pool
        
# Load database files in parallel in separate OS processes.
filenames = [
    'tests/files/foobar.dbc',
    'tests/files/foobar.dbc',
    'tests/files/padding_bit_order.dbc'
]

def load_file(filename):
    return cantools.database.load_file(filename)

with Pool(3) as pool:
    databases = pool.map(load_file, filenames)

# Add all messages to a single database.
total_database = cantools.database.Database()

for database in databases:
    total_database.messages.extend(database.messages)

total_database.refresh()

print(total_database)
```

Maybe we can later implement `__iadd__` in the Database class to allow adding one database to another, something like `total_database += database`, you get the idea.

eerimoq, 2018-09-04:
@pevsonic 

To improve performance even further a cache of pickled databases can be used. Loading a pickled database is a lot faster than loading a database using cantools.

Here is the example output using a pretty big file. Notice that the cache load time is a lot shorter than the input load time.

```
$ python3 test.py
Loaded input file.
Execution time: 2.63 s
$ python3 test.py
Loaded cached pickled file.
Execution time: 0.05 s
```

``` python
import cantools
import pickle
import time

FILENAME_INPUT_DBC = '../../a-big-file.dbc'
FILENAME_CACHED_DBC = 'a-big-file.dbc'
FILENAME_CACHED_PKL = 'a-big-file.pkl'

def load_cached_database(filename_input_dbc,
                         filename_cached_dbc,
                         filename_cached_pkl):
    """Load pickled file if it exists and contains the correct database.

    """

    with open(filename_input_dbc, 'rb') as fin:
        input_dbc = fin.read()

    with open(filename_cached_dbc, 'rb') as fin:
        cached_dbc = fin.read()

    if input_dbc != cached_dbc:
        raise ValueError()

    with open(filename_cached_pkl, 'rb') as fin:
        database = pickle.loads(fin.read())

    print('Loaded cached pickled file.')

    return database

def cache_database(database,
                   filename_input_dbc,
                   filename_cached_dbc,
                   filename_cached_pkl):
    """Add given database to the cache.

    """

    with open(filename_input_dbc, 'rb') as fin:
        input_dbc = fin.read()

    with open(filename_cached_dbc, 'wb') as fout:
        pass

    with open(filename_cached_pkl, 'wb') as fout:
        fout.write(pickle.dumps(database))

    with open(filename_cached_dbc, 'wb') as fout:
        fout.write(input_dbc)

    print('Loaded input file.')

    return database

def main():
    start_time = time.time()

    try:
        database = load_cached_database(FILENAME_INPUT_DBC,
                                        FILENAME_CACHED_DBC,
                                        FILENAME_CACHED_PKL)
    except:
        database = cantools.database.load_file(FILENAME_INPUT_DBC)
        cache_database(database,
                       FILENAME_INPUT_DBC,
                       FILENAME_CACHED_DBC,
                       FILENAME_CACHED_PKL)

    end_time = time.time()

    print('Execution time: {} s'.format(round(end_time - start_time, 2)))

if __name__ == '__main__':
    main()
```