cantools/cantools #53: add_message is failing

BayarjavkhlanBatbaatar, 2018-07-23:
This is great tool and great work. Thank you.

I have a code which was working with 23.0.0 but is failing after that to add_message to database. I do not see any change in add_message function in database.py. What could be causing the problem? Basically what I am doing is iterating though messages and adding specific messages to new database. I am getting a message with its name but failing to add it to new database. It was working fine before and generating new database.

eerimoq, 2018-07-23:
`add_message()` was renamed to `_add_message()`, and is hence private (as leading underscore indicates).

Now, instead of adding a message with `add_message()`, append the message to `messages`, and once done, call the `refresh()` function to update the internal dictionaries used for encoding and decoding lookup.

Before:

``` python
import cantools
db = cantools.database.Database()
message = cantools.database.Message(...)
db.add_message(message)
... add more messages ...
```

After:

``` python
import cantools
db = cantools.database.Database()
message = cantools.database.Message(...)
db.messages.append(message)
... add more messages ...
db.refresh()
```

BayarjavkhlanBatbaatar, 2018-07-23:
@eerimoq Thanks. It is working now.

closed 2018-07-23 by BayarjavkhlanBatbaatar