cantools/cantools #54: Decoding an encoded message returns different values

ilanbiala, 2018-07-31:
I've tried decoding an encoded message in both Cantools 28.6.0 and 22.0.1, and I'm seeing the same behavior where one signal's value is lost somehow.

```python
>>> msg.decode(msg.encode({ 'signal1': -250.0, 'signal2': -250.0, 'signal3': 1, 'signal4': 2, 'signal5': 3, 'signal6': 4, 'signal7': 0 }))
{'signal1': -250.0, 'signal2': -250.0, 'signal3': 1, 'signal4': 0, 'signal5': 3, 'signal6': 4, 'signal7': 0}
```

I noticed that https://github.com/eerimoq/cantools/blob/master/tests/test_database.py contains quite a few instances of checking whether the decoded message matches the original message passed through something equivalent to `msg.decode(msg.encode(msg_payload))`, so I'm unsure where the issue lies. All of my signals in this CAN message are big-endian, and I also noticed this weird inconsistency when printing the start and length values for my signals in this message:
```python
>>> sorted([(sig.start, sig.length) for sig in msg.signals], key=lambda sig: sig[0])
[(6, 15), (22, 15), (35, 4), (39, 4), (41, 2), (55, 8), (59, 3)]
```

I took a look at the message dump, which looks fine however: https://pastebin.com/0UufZbuj

Please let me know if more information is necessary.

eerimoq, 2018-07-31:
Are there any scaling or offsets for any of the signals?

Please provide the encoded message.

ilanbiala, 2018-07-31:
Here are (start, length, offset, and scale) tuples for each of the signals in the message.
```python
>>> sorted([(sig.start, sig.length, sig.offset, sig.scale) for sig in msg1ab.signals], key=lambda sig: sig[0])
[(6, 15, -1600, 0.1), (22, 15, -1600, 0.1), (35, 4, 0, 10), (39, 4, 0, 1), (41, 2, 0, 1), (55, 8, 0, 1), (59, 3, 0, 1)]
```

The encoded message returned is
```python
>>> msg.encode({ 'signal1': -250.0, 'signal2': -250.0, 'signal3': 1, 'signal4': 2, 'signal5': 3, 'signal6': 4, 'signal7': 0 })
b'4\xbc4\xbc\x10\x03\x04\x00'
```

ilanbiala, 2018-07-31:
Also, I noticed when looking at the array above that the value for signal1 is calculated correctly as `0x34BC`, but then it's written starting at the 0th bit. It seems like the signals aren't even being written at the correct starting points in this particular CAN message, since `0x34BC << 6` is `0xD2F00`.

eerimoq, 2018-08-01:
The big endianness start bit numbering is counter intuitive. See http://cantools.readthedocs.io/en/latest/#cantools.database.can.Signal for details. Bit 0 is bit 7 and so on.

signal4 is decoded as 0 when encoded as 2 because the scaling is 10, which when encoding gives 2 // 10 = 0, and when decoding 0 * 10 = 0. That is, the value in the encoded message is 0, and will be converted to 0 regardless of which scaling is multiplied with it when decoding. You have to encode a value that is a multiple of 10 to get a matching decoded value.

ilanbiala, 2018-08-01:
This is a partial dump of that message actually:
```
           7   6   5   4   3   2   1   0
         +---+---+---+---+---+---+---+---+
       0 |   |<--------------------------|
         +---+---+---+---+---+---+---+---+
       1 |------------------------------x|
         +---+---+---+---+---+---+---+---+
                                       +-- signal1
```
So it makes sense how 0x34BC is directly in the message actually. Thanks for the clarifications!

closed 2018-08-01 by ilanbiala