cantools/cantools #55: KCD should match labels / enums against raw signal value, not scaled

maxchill, 2018-08-27:
I created a unit test to demonstrate that KCD labels (enums) aren't always being matched against raw value, as it is specified in the KCD schema.
https://github.com/maxchill/cantools/tree/bugfix/enum_scaling

I included a commented out potential fix, however I'm not sure what the expected DBC enum matching behavior is (scaled vs raw).  I tried searching around for details but couldn't find any hard answers. Short of testing out Vector tools behavior, or comparing with other code bases I'm not sure if my proposed change fixes, or breaks DBC decoding.

Here is an excerpt from KCD schema
```
<xs:element name="Label">
    <xs:annotation>
      <xs:documentation>Descriptive name for a single value e.g. to describe an enumeration, mark special,invalid or error values.</xs:documentation>
    </xs:annotation>
    <xs:complexType>
      <xs:complexContent>
        <xs:extension base="BasicLabelType">
          <xs:attribute name="value" type="xs:nonNegativeInteger">
            <xs:annotation>
              <xs:documentation>Signal raw value that is described here.</xs:documentation>
            </xs:annotation>
          </xs:attribute>
        </xs:extension>
      </xs:complexContent>
    </xs:complexType>
  </xs:element>

  <xs:element name="LabelGroup">
    <xs:annotation>
      <xs:documentation>Descriptive name for a sequence of adjacent values.</xs:documentation>
    </xs:annotation>
    <xs:complexType>
      <xs:complexContent>
        <xs:extension base="BasicLabelType">
          <xs:attribute name="from" use="required" type="xs:nonNegativeInteger">
            <xs:annotation>
              <xs:documentation>Signal raw value the label group is starting with.</xs:documentation>
            </xs:annotation>
          </xs:attribute>
          <xs:attribute name="to" use="required" type="xs:nonNegativeInteger">
            <xs:annotation>
              <xs:documentation>Signal raw value the label group is ending with.</xs:documentation>
            </xs:annotation>
          </xs:attribute>
        </xs:extension>
      </xs:complexContent>
    </xs:complexType>
  </xs:element>
```

eerimoq, 2018-08-27:
Hello,

this looks like a bug that should be fixed. I suggest the implementation below, as it also scales values if the value has no label. Would be great if you can submit a PR with the fix.

Thanks.

``` python
def _encode_field(field, data, scaling):
    value = data[field.name]

    if isinstance(value, str):
        return field.choice_string_to_number(value)
    elif scaling:
        if field.is_float:
            return (value - field.offset) / field.scale
        else:
            value = (Decimal(value) - Decimal(field.offset)) / Decimal(field.scale)

            return value.to_integral()
    else:
        return value

def _decode_field(field, value, decode_choices, scaling):
    if decode_choices:
        try:
            return field.choices[value]
        except (KeyError, TypeError):
            pass

    if scaling:
        return (field.scale * value + field.offset)
    else:
        return value
```

eerimoq, 2018-08-27:
Yes, move the KCD definitions to a file in the tests/files folder.

closed 2018-08-29 by eerimoq