cantools/cantools #58: Multiple DBC read to same database object

zorce, 2018-09-05:
Tries to read multiple non conflicting DBC files into the same database object.
Guess this should be possible?

`dbfiles = ["file1.dbc", "file2.dbc"]

db = cantools.db.load_file(dbfiles[0])

 for file in dbfiles[1:]:
     db.add_dbc_file(file)`

zorce, 2018-09-05:
Forgot to actually say whats happening.. get some decode error:
Traceback (most recent call last):
  File "proto.py", line 256, in load_configuration
    db.add_dbc_file(file)
  File "/home/dennis/.local/lib/python3.6/site-packages/cantools/db/database.py", line 106, in add_dbc_file
    self.add_dbc(fin)
  File "/home/dennis/.local/lib/python3.6/site-packages/cantools/db/database.py", line 94, in add_dbc
    self.add_dbc_string(fp.read())
  File "/usr/lib/python3.6/codecs.py", line 321, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa1 in position 2452: invalid start byte

Worth mentioning is that each file works to load separately if I dont use the db.add_dbc_file() function

eerimoq, 2018-09-05:
Try with the newst version of cantools: `sudo pip3 install -U cantools`

zorce, 2018-09-05:
Just did and I then got some errors in the dbc file it didn't like, after removing those it seams to work ok.
The things it didn't like in the database was signals that was not connected yet so it was connected to a dummy message VECTOR__INDEPENDENT_SIG_MSG that dosen't have a size.

cantools.database.errors.Error: The signal <my_signalname> does not fit in message VECTOR__INDEPENDENT_SIG_MSG

Thank you for an awesome library!

eerimoq, 2018-09-05:
You can give `strict=False` to `load_file()` to skip various checks, among the the size check. If you are lucky your original DBC-files does not have to be modified at all.

I'm glad you like the library =)

closed 2018-09-05 by eerimoq