cantools/cantools #63: Setting 'strict' flag in 'False' does not work for overlapped signals

dmyger, 2018-10-09:
With package version 28.8.0 when trying loading dbc file having the overlapped signals message:
```
db = cantools.db.load_file(db_path, strict=False)
```
It produces exception:
```
  File "build/bdist.linux-x86_64/egg/cantools/database/__init__.py", line 112, in load_file
  File "build/bdist.linux-x86_64/egg/cantools/database/__init__.py", line 149, in load
  File "build/bdist.linux-x86_64/egg/cantools/database/__init__.py", line 204, in load_string
  File "build/bdist.linux-x86_64/egg/cantools/database/__init__.py", line 194, in load_can_database
  File "build/bdist.linux-x86_64/egg/cantools/database/can/database.py", line 130, in add_dbc_string
  File "build/bdist.linux-x86_64/egg/cantools/database/can/formats/dbc.py", line 1102, in load_string
  File "build/bdist.linux-x86_64/egg/cantools/database/can/formats/dbc.py", line 962, in _load_messages
  File "build/bdist.linux-x86_64/egg/cantools/database/can/message.py", line 59, in __init__
  File "build/bdist.linux-x86_64/egg/cantools/database/can/message.py", line 830, in refresh
  File "build/bdist.linux-x86_64/egg/cantools/database/can/message.py", line 112, in _create_codec
  File "build/bdist.linux-x86_64/egg/cantools/database/utils.py", line 200, in create_encode_decode_formats
  File "build/bdist.linux-x86_64/egg/cantools/database/utils.py", line 194, in create_little
  File "python2.7/site-packages/bitstruct-4.0.0-py2.7.egg/bitstruct.py", line 462, in pack
    return CompiledFormat(fmt).pack(*args)
  File "python2.7/site-packages/bitstruct-4.0.0-py2.7.egg/bitstruct.py", line 300, in pack
    bits = self._pack_value(info, args[i], bits)
  File "python2.7/site-packages/bitstruct-4.0.0-py2.7.egg/bitstruct.py", line 257, in _pack_value
    value_bits = info.pack(value)
  File "python2.7/site-packages/bitstruct-4.0.0-py2.7.egg/bitstruct.py", line 71, in pack
    arg))
bitstruct.Error: "u64" requires 0 <= integer <= 18446744073709551615 (got 4703919738795935662272)
```
After investigation I found the message leading to failing:
```
BO_ 2566687742 AFT1PSI2: 8 Vector__XXX
 SG_ HtrRes : 40|16@1+ (0.1,0) [0|6425.5] "ohm" Vector__XXX
 SG_ MaxRes : 32|16@1+ (250,0) [0|62500] "kohm" Vector__XXX
 SG_ Temp : 16|16@1+ (0.03125,-273) [-273|1734.96875] "�C" Vector__XXX
 SG_ RegenFailedCount : 8|8@1+ (1,0) [0|250] "" Vector__XXX
 SG_ PwrSupply : 4|2@1+ (1,0) [0|3] "" Vector__XXX
 SG_ DetectionStatus : 0|4@1+ (1,0) [0|15] "" Vector__XXX
```
By the way, version 22.0.0 open such dbc file without any error.

dmyger, 2018-10-09:
It seems, that flag has inverted behavior, if set it in True there is no such exception.

eerimoq, 2018-10-09:
Just out of curiosity, are the two signals HtrRes and MaxRes supposed to be overlapping, as the X:es in the first layout below shows? Isn't HtrRes supposed to start at bit 48 instead of 40, like in the second layout?

That said, the error message should be changed to be more informative.

First layout:

```
         7   6   5   4   3   2   1   0
       +---+---+---+---+---+---+---+---+
     0 |   |   |<-----x|<-------------x|
       +---+---+---+---+---+---+---+---+
                 |       +-- DetectionStatus
                 +-- PwrSupply
       +---+---+---+---+---+---+---+---+
     1 |<-----------------------------x|
       +---+---+---+---+---+---+---+---+
         +-- RegenFailedCount
       +---+---+---+---+---+---+---+---+
     2 |------------------------------x|
       +---+---+---+---+---+---+---+---+
 B   3 |<------------------------------|
 y     +---+---+---+---+---+---+---+---+
 t       +-- Temp
 e     +---+---+---+---+---+---+---+---+
     4 |------------------------------x|
       +---+---+---+---+---+---+---+---+
     5 |XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX|
       +---+---+---+---+---+---+---+---+
         +-- MaxRes
       +---+---+---+---+---+---+---+---+
     6 |<------------------------------|
       +---+---+---+---+---+---+---+---+
         +-- HtrRes
       +---+---+---+---+---+---+---+---+
     7 |   |   |   |   |   |   |   |   |
       +---+---+---+---+---+---+---+---+
```

Second layout:

```
         7   6   5   4   3   2   1   0
       +---+---+---+---+---+---+---+---+
     0 |   |   |<-----x|<-------------x|
       +---+---+---+---+---+---+---+---+
                 |       +-- DetectionStatus
                 +-- PwrSupply
       +---+---+---+---+---+---+---+---+
     1 |<-----------------------------x|
       +---+---+---+---+---+---+---+---+
         +-- RegenFailedCount
       +---+---+---+---+---+---+---+---+
     2 |------------------------------x|
 B     +---+---+---+---+---+---+---+---+
 y   3 |<------------------------------|
 t     +---+---+---+---+---+---+---+---+
 e       +-- Temp
       +---+---+---+---+---+---+---+---+
     4 |------------------------------x|
       +---+---+---+---+---+---+---+---+
     5 |<------------------------------|
       +---+---+---+---+---+---+---+---+
         +-- MaxRes
       +---+---+---+---+---+---+---+---+
     6 |------------------------------x|
       +---+---+---+---+---+---+---+---+
     7 |<------------------------------|
       +---+---+---+---+---+---+---+---+
         +-- HtrRes
```

eerimoq, 2018-10-09:
Fixed in 28.8.1, available on PyPI.

`strict=False` does no longer raise an exception. `script=True` does.

dmyger, 2018-10-10:
You are right, that not multiplexed overlapping signals does not work correctly. It's issue of message's description in dbc.
By the way, with `strict` flag we have two opportunity: 1) catch exception and don't load data base at all; 2) load dbc, but without information about bad messages.
Is the possibility to load file and get list of wrong messages? It allow in run time decide: ignore them if such messages not used in work, or raise exception farther.

eerimoq, 2018-10-10:
It's certainly possible to add the list of messages with overlapping signals to the exception, but not in the general case. For example, if the parser fails because one message has invalid DBC-syntax, the parser itself will fail and no recovery is possible. All messages will be discarded. `strict=False` will not matter in that case. Message errors detected after the parser succeeds should be possible to include in the list.

Is it common that there are bad messages in DBC-files? Is it the cantools implementation that is incomplete/wrong/too picky?

juleq, 2018-10-10:
„cantools too picky“: Good one :). CanDB++ will straight up crash or error out with no useful information. I have not tested this very case, though.

eerimoq, 2018-10-11:
I'll not implement it now.

closed 2018-10-11 by eerimoq