cantools/cantools #66: It's not possibile to load_database (format cdd)

leila85, 2018-11-09:
Hello guys,

I'm trying to open and parse CDD but I get this error
  File "C:\Users\gaia.littarru\AppData\Local\Programs\Python\Python37\lib\site-packages\cantools\database\__init__.py", line 228, in load_string
    raise UnsupportedDatabaseFormatError(e_dbc, e_kcd, e_sym, e_cdd)
cantools.database.UnsupportedDatabaseFormatError: CDD: "invalid literal for int() with base 10: '2,0,0'"

It seems there is a problem with data type. 
Have you already noticed this issue?

Thanks for the help,

Gaia

eerimoq, 2018-11-09:
No. If you provide a CDD-file that fails as above I can have a look at it and hopefully add support for it.

eerimoq, 2018-11-13:
Closing in a few days if there is no answer.

closed 2018-11-14 by eerimoq