cantools/cantools #67: Support for DBC double type signals

karlding, 2018-11-12:
Hi,

I was playing around with Kvaser Database Editor and realized that DBC files can contain `double` types. This is done via:

```
SIG_VALTYPE_ <can-id> <signame> : 2;
```

It seems like cantools is is currently treating this as an `(u)int64_t` instead of a `double`, which leads to incorrect C `struct` member types being generated.

Here's a sample DBC I've created

```
VERSION "HIPBNYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYYY/4/%%%/4/'%**4YYY///"

NS_ : 
	NS_DESC_
	CM_
	BA_DEF_
	BA_
	VAL_
	CAT_DEF_
	CAT_
	FILTER
	BA_DEF_DEF_
	EV_DATA_
	ENVVAR_DATA_
	SGTYPE_
	SGTYPE_VAL_
	BA_DEF_SGTYPE_
	BA_SGTYPE_
	SIG_TYPE_REF_
	VAL_TABLE_
	SIG_GROUP_
	SIG_VALTYPE_
	SIGTYPE_VALTYPE_

BS_:

BU_: TestNode

BO_ 1024 NewMessage0: 8 TestNode
 SG_ NewSignal0 : 0|64@1- (1,0) [0|0] "" Vector__XXX

CM_ BU_ TestNode "";
SIG_VALTYPE_ 1024 NewSignal0 : 2;
```

Is this something that would be supported?

Thanks!

eerimoq, 2018-11-12:
Yes, please change the code to `... in [1, 2]` instead of `== 1`.

Something similar is already done in the KCD-parser:

https://github.com/eerimoq/cantools/blob/master/cantools/database/can/formats/kcd.py#L92

eerimoq, 2018-11-12:
I added support for it in version 28.15.0.

closed 2018-11-12 by eerimoq

karlding, 2018-11-13:
Cool, thanks! I was going to do this after I got off work, but looks like you beat me to it. :joy: