cantools/cantools #69: Only max 32 characters shown in Signal Name (.dbc)

jehoff, 2018-11-20:
Hi,
I like cantools a lot, but I got a problem when I want to use it with my .dbc file.
In this dbc files I got signal names longer than 32 characters. Vector allowes signal names longer than 32 characters. But these longer names are only accesible through a special attribute.
For example: 
When i want to look for this signal name:
"FAILURE_ZELL_UEBERSPANNUNG_WARNUNG_IDX_01"
I cant find it with cantools.
The signal name is stored as:
"FAILURE_ZELL_UEBERSPANNUNG__0024"
This is because of the 32 character limitation.

I think when you interpret the .dbc file with cantools it only looks at the attribute SG_ 

[image]

But when you got longer signal names the full name ist stored in BA_ "SystemSignalLongSymbol"

[image]

Is it possible to fix this?

eerimoq, 2018-11-20:
Certainly, I'll have a look at it tonight.

eerimoq, 2018-11-20:
From version 29.0.0 message, signal and node names are long names if given in the DBC-file. Dump functionality ignores name lengths, which results in names longer than 32 characters. That is to be implemented later. I hope you are only loading DBC-files, and not dumping them. =)

jehoff, 2018-11-21:
It is working like a charm!
Thank you for your super fast reply and fix! :+1: