cantools/cantools #70: Ask for collaboration

polybassa, 2018-11-21:
Hi @eerimoq,
I really like your project cantools. 
I'm working a lot on CAN related tools, as well. In the past, I started to contribute to the Scapy project to support various automotive protocols, like UDS, ISOTP, OBD-II, GMLAN and DoIP. 
You can have a look to my work here: 
https://scapy.readthedocs.io/en/latest/advanced_usage.html#automotive-usage  
and here
https://github.com/secdev/scapy/tree/master/scapy/contrib/automotive

We also want to integrate the DBC file format and the j1939 protocol, in the future. 
Since you did all of this things in the past, I would like to ask you, if you want to 
join us and help us to make Scapy ready for the automotive world. 

Since Scapy is all about packets and packet manipulation, I think, it's the perfect project
to provide an open source tool for automotive networks.

Have a great day,
Nils

eerimoq, 2018-11-21:
Hello Nils,

Scapy looks like an interesting project, but I'm afraid I'll have to decline the invitation to join it as I'm working on too many projects already.

I wish you the best of luck making Scapy ready for the automotive world! =)

polybassa, 2018-11-21:
Thanks for your fast response. No problem. I completely understand that.

closed 2018-11-21 by polybassa