cantools/cantools #74: Feedback for c source generation. MISRA 2012 linting results.

juleq, 2018-11-27:
[linting_output.txt](https://github.com/eerimoq/cantools/files/2619446/linting_output.txt)
Hi.

Great to see cantools progressing with promising features like the c source generator. I cannot contribute myself right now but since I have access to a linter with MISRA 2012 rules set up, I thought I might at least provide its feedback to you. 

I generated from vehicle.dbc in 29.5.0 and linted using PC Lint.

The output is as redundant as you would expect, but still, there are a few pieces of interest in here.

If you are not interested right now, you can just close the issue again.

Best regards.

eerimoq, 2018-11-27:
Hello,

just out of curiosity, which warnings do you find intresting? =)

juleq, 2018-11-27:
The clashing ids might produce unexpected results depending on the toolchain you use.

Use of int is advised against.

Implicit conversions to boolean.

We are required to document every single deviation for declaration of compliance. The aim in zero deviations. We do have some in performance critical places (e.g. CAN ISR).

juleq, 2018-11-27:
Cast to unsigned:    
dst_p->gyro_rate_roll |= ((uint16_t)(src_p[7] & 0xff) << 8);
But:
int16_t gyro_rate_roll;

juleq, 2018-11-27:
Usage of | on signed types.

eerimoq, 2018-11-27:
Wow, nice feedback =)

Let's leave this issue open as some warnings should probably be fixed.

juleq, 2018-11-27:
As an example. This produces 17 Infos/Notes (MISRA Required):
    dst_p->gyro_rate_roll |= ((uint16_t)(src_p[6] & 0xff) >> 0);
    dst_p->gyro_rate_roll |= ((uint16_t)(src_p[7] & 0xff) << 8);

And this zero:
    uint16_t tmp = 0u;
    tmp |=  (uint16_t)src_p[6] & 0xffu;
    tmp |= ((uint16_t)src_p[7] & 0xffu) << 8;
    dst_p->gyro_rate_roll = (int16_t)tmp;

If you wonder about this tmp variable: The rules discourage casting complex expressions. 

This is from the (currently offline) MISRA forum:

> Rule 10.1 of MISRA C:2004 prohibits implicit conversions for signed to unsigned and vice versa

> There are no restrictions on explicit casts between simple signed and unsigned expressions ( and vice versa ). For example:

>    UINT_8 u8 = ....
>    SINT_8 s8 = ....
>    u8 = s8;            /* violates 10.1 */
>    u8 = (UINT_8)s8;   /* MISRA compliant */
> 
> Casting a complex expression requires the addition of an extra line.

>    SINT_8 stemp;  
>    u8 = (UINT_8)(s8 + s8);    /* violates 10.3 */
>    stemp = s8 + s8;
>    u8 = (UINT_8)stemp;       /*  MISRA compliant */

> Posted by and on behalf of
> the MISRA C Working Group

juleq, 2018-11-27:
Just for information. The MISRA C:2012 rule set is divided into thre categories:

- Advisory: Compliance optional. Deviations optional.
- Required: Compliance required unless justified by deviation.
- Mandatory: Compliance required always.

The aforementioned example had the 17 hits in the Required category, meaning the deviation process needs to be invoked on them.

eerimoq, 2018-11-27:
Ohh well, I'm open to fix warnings as long as the code is still clean and easy to read.

eerimoq, 2018-11-28:
Does [this](https://github.com/eerimoq/cantools/files/2623389/motohawk.txt) produce less warnings in the encode and decode functions?

juleq, 2018-11-30:
[motohawk.lint.txt](https://github.com/eerimoq/cantools/files/2634697/motohawk.lint.txt)
[motohawk.orig.lint.txt](https://github.com/eerimoq/cantools/files/2634698/motohawk.orig.lint.txt)

Indeed. I have generated using the 29.5.0 version (orig infix) and, using the same header, your new version.

eerimoq, 2018-12-01:
I made some updates to the code generator that should reduce the number of warnings. Hopefully all warnings of bitwise operations on signed numbers are gone.

karlding, 2018-12-01:
It might be worth trying to enable `-Wconversion`, which should tell you if there's any implicit conversions still occurring?

eerimoq, 2018-12-01:
The list of warnings with `-Wconversion` enbaled is long, very long. =) I'll have a look at them.

eerimoq, 2018-12-04:
@juleq The generated C code looks quite different now. It has less compiler warnings, which hopefully means less MISRA warnings. Could you run your tool again on motohawk.[hc] on master?

juleq, 2018-12-04:
@eerimoq Sorry, a little short on spare time ATM. Need assistance. How do I invoke the code generation on master? I have 30.0.0 installed and that provides the CLI. What do I do to use the masters CLI?

Here is the 30.0.0 output: [motohawk_30_0_0_lint.txt](https://github.com/eerimoq/cantools/files/2646058/motohawk_30_0_0_lint.txt)

eerimoq, 2018-12-04:
```
$ pwd
/home/erik/cantools
$ python3 -m cantools generate_c_source tests/files/motohawk.dbc 
Successfully generated motohawk.h and motohawk.c.
$ 
```

juleq, 2018-12-04:
Ahh. Nice.
[motohawk_master_lint.txt](https://github.com/eerimoq/cantools/files/2646135/motohawk_master_lint.txt)

eerimoq, 2018-12-04:
It would be great if you can make the required changes to remove more warnings as you have access to the linter. Either by manually modifying `motohawk.[hc]` and attach them here, or make required changes to `c_source.py` and submit a PR.

Warnings like `motohawk.h(45,8): Warning 621: Identifier clash ...` will not likely be fixed. Maybe optionally enable unique identifiers by giving a flag to `generate_c_source`, but before that a naming scheme has to be defined.

juleq, 2018-12-04:
How about:
[motohawk_manual_lint.txt](https://github.com/eerimoq/cantools/files/2646507/motohawk_manual_lint.txt)
[motohawk.h.txt](https://github.com/eerimoq/cantools/files/2646510/motohawk.h.txt)
[motohawk.c.txt](https://github.com/eerimoq/cantools/files/2646512/motohawk.c.txt)

I have not fixed the function like macro since I was not sure about its purpose. MISRA discourages any function like macro, but you can see deviations from that in performance critical places.

I am not sure about that decode function. It was type int, I have made it int16_t. The encode one uses ssize_t but I quess for the decode one thats just the return code?

I have not tested the code, just linted.

eerimoq, 2018-12-06:
Thanks, I might look at this at some point. Not right now.

juleq, 2018-12-06:
You already did most of it. Thanks :).

closed 2019-02-14 by eerimoq