cantools/cantools #76: Compiling C generated code test cases with more warnings enabled?

karlding, 2018-11-29:
In my projects, I typically, at a bare minimum, enable `-Wall -Werror -Wextra -Wpedantic`. I suspect I'm not the only person who does so, and GCC/Clang has a bunch of additional [useful flags](https://kristerw.blogspot.com/2017/09/useful-gcc-warning-options-not-enabled.html) as well.

Perhaps it would be good to work towards enabling these additional flags in the tests, so that it is easy for other people to include the generated C code in their projects (ie. make the code compile under the most stringent options, so that integrating will always have either equal or less-stringent options)? Currently the test files are only compiled with `-Wall -Werror -Wpedantic`.

One way to do this is starting with enabling all these flags and ignoring warnings (via `-Wno-*` flags) that the generated C code doesn't comply against so that CI is passing. Then ignored warnings can be gradually dropped as the code is made compliant, so that it doesn't have to happen all at once, since I suspect that this might be a pretty significant effort and it may make more sense to do this incrementally. Thoughts?

eerimoq, 2018-11-29:
Sounds like a good plan. I naively thought -Wall and -Wpedantic enabled all relevant warnings, but as there are more flags enabling other useful warnings those should be added to the makefile as well.

Feel free to submit a PR with the changes on your temporary branch and I'll merge it.

eerimoq, 2018-11-29:
I fixed the type limits warnings. Let me know when/if you are working on something so we don't make the same changes.

eerimoq, 2018-11-29:
I fixed the unused parameter warnings as well.

karlding, 2018-11-30:
Sounds good!

One other thing is that it may be beneficial to enable optimizations to say `-O3` or something. This is usually pretty good about finding instances where undefined behaviour is relied upon.

eerimoq, 2018-12-01:
Let's only introduce the dependency to a new version of GCC on Travis as many developers do not know how to install a newer version of GCC on their personal machine.

We could create another make target, say `test-c-gcc-8`, which is a copy of `test-c`, but with more flags. Possibly create a `SRC` make variable with all sources and use that in both make targets.

karlding, 2018-12-01:
I'll attempt to take a stab at taming `-Wconversion` (or at least a subset via `-Wsign-conversion`) later today, unless you get to it first.

eerimoq, 2018-12-01:
Ok, I'll leave it to you. Hopefully the generated code is still human readable after fixing all relevant warnings. =)

eerimoq, 2018-12-02:
An alternative is to introduce helper functions, to keep the message encode and decode functions shorter.

I have not tried to compile the code below, but it should work, right? =)

``` c
// ...

static inline uint8_t encode_left_shift_u8(uint8_t decoded,
                                           int shift,
                                           uint8_t mask)
{
    return ((uint8_t)((uint8_t)(decoded << shift) & mask));
}

static inline uint8_t decode_right_shift_u8(uint8_t value,
                                            uint8_t mask,
                                            int shift)
{
    return ((uint8_t)(encoded & mask) >> shift);
}

// ...

ssize_t multiplex_2_extended_types_encode(
    uint8_t *dst_p,
    const struct multiplex_2_extended_types_t *src_p,
    size_t size)
{
    uint32_t s10;
    uint32_t s9;
    uint8_t s0;

    if (size < 8) {
        return (-EINVAL);
    }

    memset(&dst_p[0], 0, 8);

    // Introducing functions to encode/decode singals.
    dst_p[0] |= encode_left_shift_u8(src_p->s11, 0, 0x1fu);

    switch (src_p->s11) {

    case 3:
        s0 = (uint8_t)src_p->s0;
        // Same thing here
        dst_p[1] |= encode_left_shift_u8(s0, 0, 0x0fu);

        switch (src_p->s0) {

        case 0:
            memcpy(&s10, &src_p->s10, sizeof(s10));
            dst_p[2] |= (uint8_t)((uint8_t)(s10 << 0) & 0xffu);
            dst_p[3] |= (uint8_t)((uint8_t)(s10 >> 8) & 0xffu);
            dst_p[4] |= (uint8_t)((uint8_t)(s10 >> 16) & 0xffu);
            dst_p[5] |= (uint8_t)((uint8_t)(s10 >> 24) & 0xffu);
            break;

        default:
            break;
        }
        break;

    case 5:
        memcpy(&s9, &src_p->s9, sizeof(s9));
        dst_p[3] |= (uint8_t)((uint8_t)(s9 << 0) & 0xffu);
        dst_p[4] |= (uint8_t)((uint8_t)(s9 >> 8) & 0xffu);
        dst_p[5] |= (uint8_t)((uint8_t)(s9 >> 16) & 0xffu);
        dst_p[6] |= (uint8_t)((uint8_t)(s9 >> 24) & 0xffu);
        break;

    default:
        break;
    }

    return (8);
}

int multiplex_2_extended_types_decode(
    struct multiplex_2_extended_types_t *dst_p,
    const uint8_t *src_p,
    size_t size)
{
    uint32_t s10;
    uint32_t s9;
    uint8_t s0;

    if (size < 8) {
        return (-EINVAL);
    }

    memset(dst_p, 0, sizeof(*dst_p));

    // Same thing here
    dst_p->s11 |= decode_right_shift_u8(src_p[0], 0x1fu, 0);

    switch (dst_p->s11) {

    case 3:
        s0 = 0u;
        // Same thing here
        s0 |= decode_right_shift_u8(src_p[1], 0x0fu, 0);

        if ((s0 & (1u << 3)) != 0u) {
            s0 |= 0xf0u;
        }

        dst_p->s0 = (int8_t)s0;

        switch (dst_p->s0) {

        case 0:
            s10 = 0u;
            s10 |= ((uint32_t)(src_p[2] & 0xffu) >> 0);
            s10 |= ((uint32_t)(src_p[3] & 0xffu) << 8);
            s10 |= ((uint32_t)(src_p[4] & 0xffu) << 16);
            s10 |= ((uint32_t)(src_p[5] & 0xffu) << 24);
            memcpy(&dst_p->s10, &s10, sizeof(dst_p->s10));
            break;

        default:
            break;
        }
        break;

    case 5:
        s9 = 0u;
        s9 |= ((uint32_t)(src_p[3] & 0xffu) >> 0);
        s9 |= ((uint32_t)(src_p[4] & 0xffu) << 8);
        s9 |= ((uint32_t)(src_p[5] & 0xffu) << 16);
        s9 |= ((uint32_t)(src_p[6] & 0xffu) << 24);
        memcpy(&dst_p->s9, &s9, sizeof(dst_p->s9));
        break;

    default:
        break;
    }

    return (0);
}

// ...
```

eerimoq, 2018-12-02:
Just to let you know, I pushed the refactoring of c_source.py to master.

eerimoq, 2018-12-05:
Should we close this issue? Lot's of warnings have been enabled, and more can be enabled without this issue. =)

karlding, 2018-12-05:
Sure!

closed 2018-12-05 by karlding