cantools/cantools #79: Encoding defaults to UTF-8 for dbc. Vector uses CP1252.

juleq, 2018-11-30:
Hi.

I just noticed that cantools will default to UTF-8 if the user does not pass the encoding argument, regardless of the database type (e.g. 'dbc', detected or passed by the user). I feel that this inconvenient since one could safely assume the encoding to be CP1252 for dbc files. Other encodings for the dbc file format are the exception.

I would suggest database file type specific defaults. If that is not an option, adding the encoding argument to the dbc reading related examples in the docs would present a less intrusive approach to take that off the users plate.

What do you think?

eerimoq, 2018-11-30:
Sounds like a good idea, but how should it be implemented? Choose encoding based on the file type?

juleq, 2018-11-30:
I would say:
Encoding defaults to None.

If encoding is none, select the default encoding by the type argument. If that is none, too, select the encoding by file name extension.

The selections are in a set [type, default_encoding].

The described selection process could be a local function named e.g. guess_encoding(filename, type).

eerimoq, 2018-12-01:
What if the user want to pass encoding as ``None`` to ``open()`` to use the platform dependent encoding?

https://docs.python.org/3/library/functions.html#open

Maybe adding a special default encoding, like ``'auto'``, which will do as you suggested.

juleq, 2018-12-01:
I would suggest using a special option 'platform' because I would argue that selecting the encoding based on the platform is the most edgiest of cases. The file formats are associated with specific tools that use one specific encoding. If I deviate from that, I should know what I am doing and make that explicit by using respective arguments to cantools.

I know that this strategy is different from the that open uses, but open is general purpose. And thats not necessarily the most convenient even though it is the most consistent with Python. But still, this is cantools :-).

eerimoq, 2018-12-01:
All I know is that file encoding is much harder to get right than one can imagine. There are always some use case you don't think about. When I'm in the unknown I tend to implement as few restrictions as possible in the API. An additional platform argument might work, but would be nice if it's not needed.

I totally agree that DBC-files should have the same default encoding as CANdb++, we just have to figure out how to do it in a good way =)

eerimoq, 2018-12-01:
Btw, how do you know that CP1252 is the default encoding?

juleq, 2018-12-01:
At first I assumed it. Then I noticed, that the canmatrix project uses iso-8859-1 in one of its examples. So I verified by creating a dbc with an € char in it. I read that with an editor in 1252 mode and it came out fine.

juleq, 2018-12-01:
8859 and 1252 are basically the same, but M$ replaced a few control chars with printables like €.

eerimoq, 2018-12-01:
Let's implement it as you first suggested. If someone want to use the platform encoding they can always use `load()` instead of `load_file()`.

eerimoq, 2018-12-02:
I implemented the suggested behavior on master, not yet released. Please give it a try. Consult the [documentation ](https://cantools.readthedocs.io/en/latest/#cantools.database.load_file)for details.

eerimoq, 2018-12-02:
Great that it works!

Yeah, feel free to add a dump (or write) function.

closed 2018-12-06 by eerimoq