cantools/cantools #84: Doxygen comment above struct declarations

karlding, 2018-12-05:
It's a small nit, but I was trying to compile the generated files with clang's `-Wdocumentation` and noticed that the Doxygen comments for the message `struct`s isn't actually valid Doxygen.

```
In file included from tests/main.c:6:
tests/files/c_source/motohawk.h:50:5: error: '@param' command used in a comment
      that is not attached to a function declaration [-Werror,-Wdocumentation]
 * @param enable Value as on the CAN bus.
    ^~~~~
tests/files/c_source/motohawk.h:54:5: error: '@param' command used in a comment
      that is not attached to a function declaration [-Werror,-Wdocumentation]
 * @param average_radius Value as on the CAN bus.
    ^~~~~
tests/files/c_source/motohawk.h:58:5: error: '@param' command used in a comment
      that is not attached to a function declaration [-Werror,-Wdocumentation]
 * @param temperature Value as on the CAN bus.
    ^~~~~
```

I think this is because `@param` isn't actually meant to be put on `struct` attributes.

[This StackOverflow question](https://stackoverflow.com/questions/7199151/use-doxygen-to-document-members-of-a-c-structure-outside-of-the-structure-defini) gives a workaround, if you want to have the comment outside the `struct`, the way to do this is something like

```
/** @struct motohawk_example_message_t
 *  @brief Signals in message ExampleMessage.
 *  @var motohawk_example_message_t::enable
 *  Value as on the CAN bus.
 *            Range: -
 *            Scale: 1
 *            Offset: 0
 *  @var motohawk_example_message_t::average_radius
 *  Value as on the CAN bus.
 *            Range: 0..50 (0..5 m)
 *            Scale: 0.1
 *            Offset: 0
 *  @var motohawk_example_message_t::temperature
 *  Value as on the CAN bus.
 *            Range: -2048..2047 (229.52..270.47 degK)
 *            Scale: 0.01
 *            Offset: 250
 */
```

Alternatively, this can be placed inside the `struct` above each member, if you want to fix this.

eerimoq, 2018-12-05:
Hello,

I enabled `-Wdocumentation` for clang and modified the generator to output one comment per struct member. Looks much cleaner than the old big struct comment.

``` c
/**
 * Signals in message ExampleMessage.
 *
 * Example message used as template in MotoHawk models.
 *
 * All signal values are as on the CAN bus.
 */
struct motohawk_example_message_t {
    /**
     * Range: -
     * Scale: 1
     * Offset: 0
     */
    uint8_t enable;

    /**
     * Range: 0..50 (0..5 m)
     * Scale: 0.1
     * Offset: 0
     */
    uint8_t average_radius;

    /**
     * Range: -2048..2047 (229.52..270.47 degK)
     * Scale: 0.01
     * Offset: 250
     */
    int16_t temperature;
};
```

karlding, 2018-12-06:
Cool, seems like a good approach!

closed 2018-12-06 by karlding