cantools/cantools #85: C-Code generation

xR3b0rn, 2018-12-06:
The generated C code decode-function isn't accomplishing the offset and scaling calculation. I allways have to do this by myself.

eerimoq, 2018-12-06:
Signal scaling is hard. One have to decide how to handle rounding errors. Do you have any suggestions on how to implement this as you have some experiance in this field?

xR3b0rn, 2018-12-06:
For now just multiplying and adding the offset is enought for me. I'm not familiar with this problem, could you explain me the problem in more detail? Atm. I have to test a lot with orginal data, so I could collect some expirience.

But maybe you want introduce a lacy scaling routine which is just multiplying the value by the scaling, without rounding corretly?

eerimoq, 2018-12-06:
I don't have an example of a rounding error. Maybe they don't exist, but usually they do for sone special scaling. I hav enot put much thought into this, and it would be very helpful if you could help out designing this.

Please provide example code of how you would like the generated code to look.

eerimoq, 2018-12-06:
Ok, and what are the types and values of `scaling` and `offset`? Is the scaling bigger than 1? In that case, how do you handle overflow of `di_torque_motor`?

xR3b0rn, 2018-12-06:
The code above isn't 100% what I have. And the code I'm currently working on isn't where I want to be.

I think in the future I will take double values for each signal I received, then accomplishing the scaling and offset calculations to this new data type.
So I don't need to pass data type informations between APIs because I allways can assume a double value and I don't need to make concerns about overflows.

This is maybe a problem for embedded programmers with restricted memory space, but I think for OS programmers this should be ok.

xR3b0rn, 2018-12-06:
What's about speed? I don't know because we I'm talking about double values, but I think 64bit data types should be faster on x64.

xR3b0rn, 2018-12-06:
And we are talking about physical values, so I think small rounding errors are OK for the most usecases.

xR3b0rn, 2018-12-06:
Let me know if you have any concerns.

eerimoq, 2018-12-06:
The goal is to make the generated code useful in many applications.

The current code generator suites resource constrained embedded applications fairly well (I hope). Floating point values are not used at all except for signals that are explicitly floating point in the database file. Scaling and offset defines and/or functions should probably be generated as well for completeness, preferably without floating point values.

The tricky part of generating code to many applications is that each application want to have its own representation of the signals. Some want float, some double, some integers, some strings, some classes, who knows, nothing would surprise me. The generated code can not take everything into account, obviously.

I don't have any use case myself at the moment, and it's kinda hard to design an API if you don't know how the user want to use it. All I know is that you want double, which sounds like a fairly common use case. Though, I'm pretty sure signals with scaling 1 and offset 0 often are represented by an integer in many applications. Signals with choices/enumerations are probably enum or integers. All of these mixed in each message. There are lots of cases to take into account and it's very hard to generate code that is somewhat right.

I suggest that we start by generating offset/scaling functions that when decoding returns a double, and then encoding returns the database type, often integers and sometimes float.

Thoughts?

xR3b0rn, 2018-12-06:
I'm using DBC and I thought in DBC are no float values. If you want floats in DBC you have to work with scaling? I don't know any of the other supported can frame description formats.

> I suggest that we start by generating offset/scaling functions that when decoding returns a double, and then encoding returns the database type, often integers and sometimes float.

I didn't get the last one, why you differentiate between encoding and decoding? But with the first part you mean simply replacing the output struct member types with double?

If yes, I would say this solution would hit the biggest set of usecases of the users.

But I think there could be one more problem:
- big values couldn't be represent as double correctly (perhaps you meant this with: "rounding errors"). For physical values I don't see a big problem (because it's not that bad if you round a physical value) but If the ECU is transmitting a state and the state has e.g. 64bit, the value could be rounded if the value gets converted to double. So the receiver wouldn't be able to determine the state of the ECU correctly.

xR3b0rn, 2018-12-06:
Yes, I like this. Perhaps I could do this work next week.

But maybe a other naming convention for the decoding functions? E.g. message unpacking and signal decoding?

juleq, 2018-12-06:
I would recommend using different verbs than en-/decode. E.g. physical_to_raw() and raw_to_physical(). If your internal representation is double you would e.g. encode(physical_to_raw(9.81f)).

I think all double for the physical representation  would be fine because it keeps it simple and serves as example. If you need a custom integer based fixed point representation, you would write specific conversions for that. There are code generators that automate the fixed point arithmetic (Mathworks, dSpace) but I think thats a tough nut, too. It would need to be highly configurable.

eerimoq, 2018-12-06:
Message unpacking and signal decoding sounds pretty good.

eerimoq, 2018-12-07:
News in release version 32.0.0:

- Renamed `*_encode()` to `*_pack()`.
- Renamed `*_decode()` to `*_unpack()`.
- Signal encode and decode functions, applying scaling and offset. Decoded signals are of type `double`.

Let me know if anything could be improved.

juleq, 2018-12-07:
Agreed. But wasn‘t encoding used for packing before?
 
Ok, just saw your notes on 32.0.0.

xR3b0rn, 2018-12-07:
This was fast, thank you. I will test it as soon as I'm back at work.

karlding, 2018-12-07:
To add another use case data-point, currently I'm using the generated code in a project on an ARM Cortex-M0 processor, which has support for neither single nor double precision floating point. As such, introducing a `double` that isn't absolutely necessary will introduce many cycles (high hundreds if not thousands), which would be unacceptable for this use case. If supporting scaling means an introduction of `double` types, I would propose either/any combination of these, which would not introduce this performance hit:

1. Making this controlled by a command line flag that determines whether or not to generate code that performs scaling. This would only generate code to perform scaling and generate `double` `struct` members as needed (instead of the current types) if the flag is enabled/disabled.
2. Generating separate files that can be used in their own compilation unit. One file contains headers/code that performs scaling and only uses `double` types, and one that doesn't. A user can pick which implementation to use and compile against.
3. Using optional "getter" and "setter" functions to operate on `double` types, which would get optimized out by the compiler if unused. So physical values are still kept as-is, and then specific functions are created that take these physical values and return the scaled value. A downside is that now you either need another `struct` to encapsulate the data, or mandate in the API contract that everything needs to be accessed via these getter/setters if they want scaled values.
4. Not use `double` types if scaling values are not set in the input database. Not _really_ a fan of this due to the seemingly "magic" behaviour, but I'm fine if this is what you choose to go with, since this is basically the same as the first two suggestions.

eerimoq, 2018-12-07:
The idea is that the message struct and the message pack and unpack functions only will contain float and/or double if the database file contains signals with these types. If the database only contains signed and unsigned signals no float or double will be present.

The signal encode and decode functions will always contain a double, but using these functions are optional to use. The core functionality to pack and unpack signals can be used without these functions.

I'll probably add a command line option to disable code generation of the signal encode and decode functions, as there are probably some compiler that will complain about them even if they are not used by the application.

Also, I'm open to generate defines and/or functions to perform scaling using fixed point numbers or something similar, allowing signal encoding and decoding without double. Ideas of how to do this are welcome.

eerimoq, 2018-12-07:
I added two command line options in version 32.1.0:

--database-name: The data base name. Generated files will be called <database-name>.[hc], and all structs, defines and functions are prefixed with <database-name> instead of the input file name.

--no-floating-point-numbers: Currently disabled code generation of signal encode and decode functions.

xR3b0rn, 2018-12-10:
It is working perfektly. Thanks!

closed 2018-12-10 by eerimoq

karlding, 2018-12-11:
Cool, I just wanted to clarify to make sure that everything wasn't being converted to a `double` if it supported scaling, since that was the impression I was getting.

I think I was getting a little confused because `pack` used to be called `encode` :joy:

dzid26, 2021-04-15:
Just to understand latest implementation.  Does cantools support single precision `float` currently? 
I keep getting double everywhere no matter the scaling or signal type `SIG_VALTYPE_`.