cantools/cantools #86: Discussion: Bit field in message struct for more straightforward packing and unpacking

yangliu28, 2018-12-10:
Hi Erik,

Cheers! Excellent job you have done, especially the generating c source code part that I found particular useful to me. When reading the generated code, a colleague of mine reminded me that using bit field in the message structs might be a good idea, and I very much like to know your opinion. I'll use the `ExampleMessage` from *motohawk.dbc* as an example.

Right now the struct definition is as below. The packing and unpacking functions are done with bit operations to set and extract data.
```c
struct motohawk_example_message_t {
    uint8_t enable;
    uint8_t average_radius;
    int16_t temperature;
};
```

If adding bit fields, the packing and unpacking can be simplified or eliminated, by force converting the incoming data to fit the struct, and return the pointer directly for outgoing data.
```c
struct motohawk_example_message_t {
    uint8_t enable: 1;
    uint8_t average_radius: 6;
    int16_t temperature: 12;
} __attribute__((packed));
```

It introduces some new issues as well. So if applicable, at least some conversions are needed.
- The bit-wise endianess in a CAN signal can be both little-endian or big-endian, but in computer system it is always big-endian.
- In my experiment, when instantiating a message from the struct, the signals seem to be stored in a reverse order compared to their sequence.
- The total size of a CAN message could be larger than the message struct. The above struct takes only 3 bytes, while it should take 8 bytes from .dbc file definition.

I am still studying the generated code, so let me know any of my misunderstandings. Do you think the bit field struct has its valid advantages here? Or it may complicate things in an unnecessary way?

Yang

eerimoq, 2018-12-10:
Hello,

I don't know the C standard very well, so I stick to bit-wise operations as I think I know something about them. That said, bit fields could be used in many cases to let the compiler very efficiently pack and unpack of signals, but I'm pretty sure the this is not portable at all. Compilers probably implement this differently, the ordering of bit fields may vary and it is probably endianness dependent.

How much of a performance improvement do you see using bit fields? And does this improvement really matter?

juleq, 2018-12-10:
My two cents:

Bitfields will save you some memory if you really need that. I cannot tell if it will also save you cycles as that heavily depends on the compiler and processor (those will do the masking for you). And you loose portability in many cases.

That said: We do use bitfields and unions for packing/unpacking CAN messages that are our own and that do not change. If you need compliance with e.g. MISRA, you need a declare a deviation to document the precautions you took to avoid common pitfalls with byte alignment and byte packing. To control alignment and packing we use (non portable) compiler pragmas.

If I would start that part over I would compare the code that my compiler emits for a few samples. I probably do not think it is worth coping with the drawbacks for a few cycles/memory nowadays.

yangliu28, 2018-12-10:
@eerimoq @juleq 
I haven't thought of whether it would be portable, this is a deciding point here. I wouldn't go for bit fields because of saving a few cycles/memory neither.

My concern actually came up when using the generated source code, we somehow misinterpreted a received CAN message. And it took a little while to trace the error, because the bit operations are less readable to bit field struct, which has message structure defined bit-wise following given definition. So it's ease of maintainability that matters in this case. But as you mentioned, this does not make a big difference in performance, aside from byte alignment issues.

It's good to know your opinions. Many thanks!

juleq, 2018-12-10:
@yangliu Good point. Readability is probably best with struct/union. I think overall it is a valid approach when everyone working with the code is aware of the hidden pitfalls.

karlding, 2018-12-11:
The only major advantage I see is in degenerate cases for code size. Currently, in the worst case, you'd have 64 1-bit signals that are assigned `uint8_t`s, which results in 7 bits wasted for each signal (for a total of 56 bytes of "wasted data"). Maybe there's a worse case with multiplexed signals? Using a bitfield will probably pack this better, at the expense of portability. I suppose you could take a command line argument that specifies the endianness of the system you wish to compile for, but if you happen to switch compilers, there's no guarantee it works the same way.

The portable approach for bitfields would be to introduce an opaque `struct`/`typedef` that wraps a `uint64_t`, and then operate directly on that, instead of generating the `struct`s as they exist today.

With that said, there's currently some low-hanging fruit in reducing the memory footprint for certain cases where "slop" is added due to alignment requirements. It's fairly trivial to re-arrange the struct members and generate code such that `struct` padding is minimized, while still being portable. I think in the worst case, you would save ~4 bytes per `struct`, which isn't a lot, but it's fairly low-effort (probably just involves iterating over the signals in sorted order by length when building the `struct`).

Regarding maintainability, I'm of the opinion that autogenerated code should be "write once, comprehensively test the generator and validate its output, and then forget about it".

Perhaps adding property-based testing (via [silentbicycle/theft](https://github.com/silentbicycle/theft)) or fuzzing (via [afl](http://lcamtuf.coredump.cx/afl/)) to the test harness would help build confidence in the generated code's correctness?

eerimoq, 2018-12-12:
I added the option `--bit-fields` to generate bit fields in message structs to save some RAM, especially if compiled with `-fpack-struct`.

closed 2018-12-12 by eerimoq

eerimoq, 2019-01-06:
@karlding Do you have any experience from [libFuzzer](https://llvm.org/docs/LibFuzzer.html)? I used it for the first time today in the [asn1tools](https://github.com/eerimoq/asn1tools) project, and I liked it. No bugs found, but it was easy to identify dead code which I removed.

The asn1tools project is similar to this project as it also generates C source code with encode and decode functions. It would be easy to use libFuzzer in cantools as well to fuzz test the unpack functions for illegal memory accesses for various "random" CAN frame data.

karlding, 2019-01-09:
No, I've heard good things about it though! I agree, it would be useful to add support and catch these classes of problems.