cantools/cantools #91: Lack of message type and signal default value information

BrushLee90, 2019-01-08:
Message or frame type and signal default value information sometimes are also needed from dbc file.
Message type, such as can can standard, can extended, can FD standard. example as following:
[image]

So, the two attribution can be added?  Thank you for hard work~

eerimoq, 2019-01-08:
Please attach a valid DBC-file with these attributes.

eerimoq, 2019-01-09:
I cannot find the message nor the signal from your first comment in the DBC file. Please write a better description of the issue with all information required to implement support for it.

IvanEscamilla, 2019-02-26:
Hi, I just made a PR that solves the problem of the signal initial value. #112

BrushLee90, 2019-03-07:
got it. thank you!

closed 2019-03-07 by BrushLee90