cantools/cantools #92: Support for Generating C Code From .sym Files With No SEND or RECEIVE Section

garethellis0, 2019-01-09:
Currently, when trying to generate C code from a `.sym` file with no `{SEND}` section (just a `{SENDRECEIVE}` section), the below error is produced. A similar error is produced if `{RECEIVE}` is missing.

Please let me know if you'd like any more information (and thanks for the awesome project!).

```
Traceback (most recent call last):
  File "/home/gareth/programming/formula_e/Consolidated-Firmware/src/shared/dbc/generate_can_c_code.py", line 25, in <module>
    generate_code_from_sym_file(sym_filename)
  File "/home/gareth/programming/formula_e/Consolidated-Firmware/src/shared/dbc/generate_can_c_code.py", line 9, in generate_code_from_sym_file
    dbase = load_file(database_name + ".sym", database_format="sym")
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/__init__.py", line 170, in load_file
    strict)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/__init__.py", line 238, in load
    strict)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/__init__.py", line 305, in load_string
    return load_can_database('sym')
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/__init__.py", line 287, in load_can_database
    db.add_sym_string(string)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/can/database.py", line 198, in add_sym_string
    database = sym.load_string(string, self._strict)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/can/formats/sym.py", line 481, in load_string
    messages = _load_messages(tokens, signals, strict)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/can/formats/sym.py", line 457, in _load_messages
    messages = _load_message_section('{SEND}', tokens, signals, strict)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/can/formats/sym.py", line 438, in _load_message_section
    for message_tokens in message_section_tokens:
TypeError: 'NoneType' object is not iterable

Process finished with exit code 1
```

eerimoq, 2019-01-09:
Hi,

the SYM format is the least tested format in cantools and there are likely more bugs or unsupported features left to implement.

I fixed this issue in release 32.4.0, available on PyPI.

garethellis0, 2019-01-10:
Awesome, thank you so much for quickly adding support for this!

eerimoq, 2019-01-10:
You're welcome! Let me know if you find more bugs and we can fix those as well.

closed 2019-01-10 by eerimoq