cantools/cantools #93: Support for Intel Format Floats in .sym Files

garethellis0, 2019-01-09:
If I generate a `.sym` file with a Intel format float, the following line is produced in `{Signals}`:
```
Sig=Signal_Float_1 float /p:10
```

This seems to cause an error when parsing the file (shown below). Please let me know if you need any more information!

```
Traceback (most recent call last):
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/textparser.py", line 834, in parse
    tokens = self.tokenize(text)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/can/formats/sym.py", line 123, in tokenize
    raise TokenizeError(string, mo.start())
textparser.TokenizeError: Invalid syntax at line 17, column 26: "Sig=Signal_Float_1 float >>!<</p:10"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/gareth/programming/formula_e/Consolidated-Firmware/src/shared/dbc/generate_can_c_code.py", line 25, in <module>
    generate_code_from_sym_file(sym_filename)
  File "/home/gareth/programming/formula_e/Consolidated-Firmware/src/shared/dbc/generate_can_c_code.py", line 9, in generate_code_from_sym_file
    dbase = load_file(database_name + ".sym", database_format="sym")
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/__init__.py", line 170, in load_file
    strict)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/__init__.py", line 238, in load
    strict)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/__init__.py", line 305, in load_string
    return load_can_database('sym')
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/__init__.py", line 287, in load_can_database
    db.add_sym_string(string)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/can/database.py", line 198, in add_sym_string
    database = sym.load_string(string, self._strict)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/cantools/database/can/formats/sym.py", line 476, in load_string
    tokens = Parser60().parse(string)
  File "/home/gareth/.local/share/virtualenvs/Consolidated-Firmware-A0WH47y7/lib/python3.5/site-packages/textparser.py", line 841, in parse
    raise ParseError(text, e.offset)
textparser.ParseError: Invalid syntax at line 17, column 26: "Sig=Signal_Float_1 float >>!<</p:10"

Process finished with exit code 1
```

eerimoq, 2019-01-09:
I fixed this issue in release 32.4.0, available on PyPI.

garethellis0, 2019-01-10:
Awesome, thank you so much for quickly adding support for this!

closed 2019-01-10 by eerimoq