cantools/cantools #96: Support For Enum Type Signals in .sym Files

garethellis0, 2019-01-11:
Symbols that have an Enum-type signal in `.sym` files (example below) produce the below error.
```
[ExampleSymbol1]
ID=0F0h
Len=8
Sig=Enum 0
```

```
Traceback (most recent call last):
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\textparser.py", line 839, in parse
    return Grammar(self.grammar()).parse(tokens, token_tree)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\textparser.py", line 640, in parse
    raise GrammarError(tokens.peek_max().offset)
textparser.GrammarError: Invalid syntax at offset 5074.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "src\shared\CanMsgs\generate_c_code_from_sym.py", line 50, in <module>
    generate_code_from_sym_file(sym_filename)
  File "src\shared\CanMsgs\generate_c_code_from_sym.py", line 32, in generate_code_from_sym_file
    dbase = load_file(database_name + ".sym", database_format="sym")
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 170, in load_file
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 238, in load
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 305, in load_string
    return load_can_database('sym')
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 287, in load_can_database
    db.add_sym_string(string)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\database.py", line 198, in add_sym_string
    database = sym.load_string(string, self._strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 490, in load_string
    tokens = Parser60().parse(string)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\textparser.py", line 841, in parse
    raise ParseError(text, e.offset)
textparser.ParseError: Invalid syntax at line 373, column 5: "Sig=>>!<<Enum 0"
```

Please let me know if you need any more information! Thank you!

eerimoq, 2019-01-11:
It would be helpful if you can attach an example .sym-file that I can use for testing.

eerimoq, 2019-01-11:
Fixed in 32.5.1, available on PyPI.

Let me know when you find more bugs.

garethellis0, 2019-01-11:
I will make sure to attach an example `.sym` file in the future. Thank you!

garethellis0, 2019-01-11:
Reopening this as it looks like that while Enum's are no longer throwing error's while being parsed, they are not being parsed entirely correctly. I believe this is because PCAN Symbol Editor (the `.sym` editor I'm using) is not using the `/e` option on signals or variables, but rather is using Enum as a type. Below is an example of a `.sym` with a symbol that contains both a Enum signal and Enum variable (both using the same enum, `Enum1`). In both cases, the resulting struct (also shown below) contains `uint8_t` values for the fields (which I believe is expected) but there are no "choices" (to use the language used in `sym.py`) generated. 

```
FormatVersion=6.0 // Do not edit this line!
Title="Untitled"

{ENUMS}
Enum=Enum1(1="asdfasdf", 0="asdfasd")

{SIGNALS}
Sig=Signed_Int_Char char
Sig=Enum_Signal Enum1 8

{SENDRECEIVE}

[Symbol1]
ID=000h
Len=8
Var=SomeEnum Enum1 40,1
Sig=Enum_Signal 41
```
```
#define CANMSGS_SYMBOL1_FRAME_ID (0x00u)

/**
 * Signals in message Symbol1.
 *
 * All signal values are as on the CAN bus.
 */
struct CanMsgs_symbol1_t {
    /**
     * Range: -
     * Scale: 1
     * Offset: 0
     */
    uint8_t some_enum;

    /**
     * Range: -
     * Scale: 1
     * Offset: 0
     */
    uint8_t enum_signal;
};
```

For your reference, starting on page 3167 of the following PDF, there is a specification for symbol files:
[PcanExpl_enu.pdf](https://github.com/eerimoq/cantools/files/2751222/PcanExpl_enu.pdf)

garethellis0, 2019-01-12:
After some more investigation, it appears that the `.sym` format is still using `/e`, but only when assigning an Enum to data type _other_ then `Enum`.  However, when I pass in the following `.sym`, where the `PDM_HeartBeat` symbol has a signed int variable with `Enum1` assigned to it, I get the below error:
```
FormatVersion=6.0 // Do not edit this line!
Title="Untitled"

{ENUMS}
Enum=Enum1()

{SIGNALS}

{SENDRECEIVE}

[PDM_HeartBeat]
ID=001h
Len=8
Var=Variable1 signed 0,8 /e:Enum1
```
```
Traceback (most recent call last):
  File "src\shared\CanMsgs\generate_c_code_from_sym.py", line 77, in <module>
    generate_code_from_sym_file(sym_filename)
  File "src\shared\CanMsgs\generate_c_code_from_sym.py", line 47, in generate_code_from_sym_file
    dbase = load_file(database_name + ".sym", database_format="sym")
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 170, in load_file
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 238, in load
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 305, in load_string
    return load_can_database('sym')
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 287, in load_can_database
    db.add_sym_string(string)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\database.py", line 198, in add_sym_string
    database = sym.load_string(string, self._strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 583, in load_string
    messages = _load_messages(tokens, signals, strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 561, in _load_messages
    messages += _load_message_section('{SENDRECEIVE}', tokens, signals, strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 552, in _load_message_section
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 507, in _load_message
    signals),
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 478, in _load_message_signals
    signals)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 430, in _load_message_signals_inner
    for variable in message_tokens[5].get('Var', [])
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 430, in <listcomp>
    for variable in message_tokens[5].get('Var', [])
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 396, in _load_message_variable
    tokens[8], enums)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 266, in _load_signal_attributes
    enum = enums[value]
KeyError: 'Enum1'
```

So perhaps there is more then one issue here?

eerimoq, 2019-01-12:
Fixed in 32.5.3, available on PyPI!

closed 2019-01-15 by eerimoq