cantools/cantools #97: .sym File Parsing Fails On Empty Enums

garethellis0, 2019-01-11:
When parsing `.sym` files, parsing fails with the below error:
```
Traceback (most recent call last):
  File "src\shared\CanMsgs\generate_c_code_from_sym.py", line 67, in <module>
    generate_code_from_sym_file(sym_filename)
  File "src\shared\CanMsgs\generate_c_code_from_sym.py", line 49, in generate_code_from_sym_file
    dbase = load_file(database_name + ".sym", database_format="sym")
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 170, in load_file
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 238, in load
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 305, in load_string
    return load_can_database('sym')
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 287, in load_can_database
    db.add_sym_string(string)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\database.py", line 198, in add_sym_string
    database = sym.load_string(string, self._strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 574, in load_string
    tokens = Parser60().parse(string)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\textparser.py", line 841, in parse
    raise ParseError(text, e.offset)
textparser.ParseError: Invalid syntax at line 5, column 12: "Enum=Enum1(>>!<<)"
```
when parsing the following `.sym` with an empty enum:
```
FormatVersion=6.0 // Do not edit this line!
Title="Untitled"

{ENUMS}
Enum=Enum1()
```
but parses just fine when there is at least one value in the enum, ex:
```
FormatVersion=6.0 // Do not edit this line!
Title="Untitled"

{ENUMS}
Enum=Enum1(0="")
```

Please let me know if you need anything else! Thanks!

eerimoq, 2019-01-11:
Thanks for the detailed bug report. Makes the troubleshooting easy. =)

Fixed in version 32.5.2, available on PyPI!

garethellis0, 2019-01-11:
Confirmed fixed in 35.5.2, thanks!

closed 2019-01-11 by garethellis0