cantools/cantools #100: Support for bit Type In .sym Files

garethellis0, 2019-01-12:
Symbols in `.sym` files can hold variables of type `bit` (example below). It would be awesome if this could be supported (currently looks like it's being ignored as a "unsupported type")! This type is particularly useful for my team's use case, as we 1-hot encode multiple errors in a single message.
```
FormatVersion=6.0 // Do not edit this line!
Title="Untitled"

{ENUMS}
Enum=Enum1()

{SENDRECEIVE}

[PDM_HeartBeat]
ID=001h
Len=8
Var=Variable1 bit 0,1
```

Please let me know if you need anything else! Thank you!

eerimoq, 2019-01-12:
Today the type system in cantools only handles signed, unsinged and float. The type system should probably be improved, say with a `type`-attribute in the signal class, but as a first step I suggest representing a bit with an unsigned value of length one.

garethellis0, 2019-01-12:
Do you mean translating a `bit` to an unsigned value of length one within `cantools`? Or just using a unsigned value of length one instead of a `bit`? I don't think the first option would be too tricky, seems to be just a matter of tweaking the parser?

eerimoq, 2019-01-12:
Yes, translate within cantools.

eerimoq, 2019-01-12:
Fixed in 32.5.3, available on PyPI!

closed 2019-01-15 by eerimoq

garethellis0, 2019-01-15:
Sorry for taking so long to get to this. Generation runs fine, but the generated variable in the struct has a range of `-`, whereas I would expect something like (0..1). Bit of a nit-pick, but just thought I'd mention it here.

eerimoq, 2019-01-15:
Makes sense. I'll have a look at it.

eerimoq, 2019-01-15:
Added in version 32.6.1.

garethellis0, 2019-01-16:
Confirmed fixed in 32.6.1!