cantools/cantools #98: Support for String Type in `.sym` Files

garethellis0, 2019-01-12:
When parsing the following `.sym` file:
```
FormatVersion=6.0 // Do not edit this line!
Title="Untitled"

{ENUMS}
Enum=Enum1()

{SIGNALS}
Sig=Signed_Int_Char char
Sig=Enum_Signal Enum1 8

{SENDRECEIVE}

[PDM_HeartBeat]
ID=001h
Len=8
Var=Variable2 string 0,64 /d:rDefaultS
```
where the `PDM_HeartBeat` symbol has a 64 bit "string" variable with default value "DefaultS", the following error is produced:
```Traceback (most recent call last):
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\textparser.py", line 839, in parse
    return Grammar(self.grammar()).parse(tokens, token_tree)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\textparser.py", line 640, in parse
    raise GrammarError(tokens.peek_max().offset)
textparser.GrammarError: Invalid syntax at offset 218.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "src\shared\CanMsgs\generate_c_code_from_sym.py", line 77, in <module>
    generate_code_from_sym_file(sym_filename)
  File "src\shared\CanMsgs\generate_c_code_from_sym.py", line 47, in generate_code_from_sym_file
    dbase = load_file(database_name + ".sym", database_format="sym")
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 170, in load_file
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 238, in load
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 305, in load_string
    return load_can_database('sym')
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 287, in load_can_database
    db.add_sym_string(string)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\database.py", line 198, in add_sym_string
    database = sym.load_string(string, self._strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 578, in load_string
    tokens = Parser60().parse(string)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\textparser.py", line 841, in parse
    raise ParseError(text, e.offset)
textparser.ParseError: Invalid syntax at line 16, column 30: "Var=Variable2 string 0,64 /d:>>!<<rDefaultS"
```

Please let me know if you need anything else! Thanks!

eerimoq, 2019-01-12:
Today the type system in cantools only handles signed, unsinged and float. The type system should probably be improved, say with a `type`-attribute in the signal class, but as a first step I suggest representing a `string` with an unsigned value of a multiple of 8 bits and `char` as an unsigned value of length 8.

eerimoq, 2019-01-12:
Fixed in 32.5.3, available on PyPI!

garethellis0, 2019-01-15:
Attempting to generate C code from the original `.sym` above seem to still result in the same error in 32.5.3? Copying it here again just in case there is a difference:
```
Traceback (most recent call last):
  File "C:\Users\Gareth\Consolidated-Firmware\src\shared\CanMsgs\generate_c_code_from_sym.py", line 79, in <module>
    generate_code_from_sym_file(sym_filename)
  File "C:\Users\Gareth\Consolidated-Firmware\src\shared\CanMsgs\generate_c_code_from_sym.py", line 48, in generate_code_from_sym_file
    dbase = load_file(database_name + ".sym", database_format="sym")
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 170, in load_file
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 238, in load
    strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 305, in load_string
    return load_can_database('sym')
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\__init__.py", line 287, in load_can_database
    db.add_sym_string(string)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\database.py", line 198, in add_sym_string
    database = sym.load_string(string, self._strict)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\cantools\database\can\formats\sym.py", line 605, in load_string
    tokens = Parser60().parse(string)
  File "C:\Users\Gareth\.virtualenvs\Consolidated-Firmware-74Z7PhuA\lib\site-packages\textparser.py", line 841, in parse
    raise ParseError(text, e.offset)
textparser.ParseError: Invalid syntax at line 16, column 30: "Var=Variable2 string 0,64 /d:>>!<<rDefaultS"
```

eerimoq, 2019-01-15:
True, I'll have a look at it.

eerimoq, 2019-01-15:
Should work in version 32.6.1.

garethellis0, 2019-01-16:
Confirmed that 32.6.1 produces the following:

```
struct CanMsgs_pdm_heart_beat_t {
    /**
     * Range: -
     * Scale: 1
     * Offset: 0
     */
    uint64_t variable2;
};
```

Thanks!

closed 2019-01-16 by garethellis0