cantools/cantools #99: Setting /max On Variable in .sym Leads To Malformed Range In Generated C Code

garethellis0, 2019-01-12:
The following `.sym` message:
```
FormatVersion=6.0 // Do not edit this line!
Title="Untitled"

{ENUMS}
Enum=Enum1()

{SIGNALS}
Sig=Signed_Int_Char char
Sig=Enum_Signal Enum1 8

{SENDRECEIVE}

[PDM_HeartBeat]
ID=001h
Len=8
Var=Variable1 unsigned 42,3 /max:1
```
produces the following `struct`:
```
/**
 * Signals in message PDM_HeartBeat.
 *
 * All signal values are as on the CAN bus.
 */
struct CanMsgs_pdm_heart_beat_t {
    /**
     * Range: ..1 (..1 -
     * Scale: 1
     * Offset: 0
     */
    uint8_t variable1;
};
```

Note that the range is ` ..1 (..1 -`, where  I would expect it to be something like `* 0..1 (0..1 -)`? Looks like this is a case of `/max` somehow being misinterpreted? What's really confusing here is that the right value somehow makes it all the way from `.sym`->`internal database`->`C source`, so I _suspect_ this is a bug with C source generation?

Please let me know if you need anything else! Thanks!

eerimoq, 2019-01-12:
Seems the PCAN SymbolEditor always displays minimum and maximum values, but only writes these to file if they put restrictions on the value. For example the minimum value is not written to file in your example above because an unsigned value cannot be less than 0.

garethellis0, 2019-01-12:
I think it's just the closing parentheses here, I was just dumping information that I found in case it was at all useful.

eerimoq, 2019-01-12:
Ok, great!

eerimoq, 2019-01-12:
Fixed in 32.5.3, available on PyPI!

closed 2019-01-15 by eerimoq

garethellis0, 2019-01-15:
Confirmed resolved in 32.5.3