cantools/cantools #103: Multiplexed messages in monitor output

karlding, 2019-01-13:
Currently, when using multiplexed messages, they get decoded under the same message like so:

```
Received: 104, Discarded: 49, Errors: 0
   TIMESTAMP  MESSAGE
       2.750  BatteryVoltageTemperature(
                  BatteryIndex_mux: 28,
                  Voltage_28: 39294,
                  Temperature_28: 23122
              )
```

When you have multiplexed messages that come at high frequency, this is somewhat hard to parse, as it updates too quickly.

I think it would be useful to either:

1. Display all the muxed values under separate messages
2. Display all the muxed values under the same message

`(1)` would look something like:

```
Received: 104, Discarded: 49, Errors: 0
   TIMESTAMP  MESSAGE
       2.750  BatteryVoltageTemperature(
                  BatteryIndex_mux: 28,
                  Voltage_28: 39294,
                  Temperature_28: 23122
              )
       
       2.754  BatteryVoltageTemperature(
                  BatteryIndex_mux: 29,
                  Voltage_29: 39477,
                  Temperature_29: 23022
              )
```

And `(2)` would look something like:

```
Received: 104, Discarded: 49, Errors: 0
   TIMESTAMP  MESSAGE
       2.754  BatteryVoltageTemperature(
                  BatteryIndex_mux: 29,
                  Voltage_28: 39294,
                  Voltage_29: 39477,
                  Temperature_28: 23122
                  Temperature_29: 23022
              )
```

The multiplexer value would denote the last MUX'd value received (or it could
optionally be hidden).

I guess this would also be applicable to the `decode` subparser as well.

What are your thoughts?

eerimoq, 2019-01-13:
In some systems it's irrelevant which multiplexer value (or even message) the signal is part of. For these systems a union of all signals for all multiplexer values would make sense. One could even argue that the message should not be displayed at all, but only the union of all signals in the system.

As I understand it you are using the multiplexer value as an identifier of the source of the voltage and temperature, so a union of all signals in the message wont work as only the latest will be displayed. Am I missing something here?

juleq, 2019-01-13:
@karlding: So you‘re in the BMS club, heh? Cheers :).

For my telemetry, I do it like @eerimoq suggests: I have update listeners attached to all my signals. On update, a signal value is copied to a signal buffer (union).

The application samples the signal buffer. 

Further, each signal has an invalidation timer attached to it. If a signal has not been updated for e.g. three consecutive times, it is considered frozen and invalidated.

For a message based approach I would argue, that a message with 3 signals and 3 mux ids should have 9 individual signals. Since you would loose timing information when only the last received mux id defines the timestamp, you could add a muxTimestamps collection to keep the information for the other mux ids. 

The application or the library could then calculate the age of a signal easily and act accordingly.

karlding, 2019-01-13:
Haha, I'm just an undergrad right now. The sample is from some stuff I'm working on a design team at school right now for a solar car. I'm looking to get rid of our one-off scripts and custom codegen/debugging stuff we've built over the course of developing our current car, since a lot of our core firmware team is graduating soon, and I suspect there's a lot of complexity and maintenance overhead we've inadvertently introduced that can somewhat be alleviated using standardized tooling (to make it easier for the new people).

Yeah, exactly. My use case is that I have one message that sends `(module index, module voltage, module temp)`. The data is currently only sent out for debug purposes, so it is only periodically sent out. Instead of using separate messages for each module index, we're using the first 2 bytes to multiplex the voltage and temperature to determine which battery module the reading corresponds to.

In PCAN-Explorer, they take the approach where they show each MUX'd signal as it's own "message" (ie. the first approach) in order to retain timing information.

eerimoq, 2019-01-14:
I prefer alternative `(1)`, but maybe without the `_28` / `_29` signal suffixes?

karlding, 2019-03-28:
Closing as the PR was merged and available as of 32.14.0

closed 2019-03-28 by karlding