cantools/cantools #105: Support For Variable Comments In .sym Files

garethellis0, 2019-01-25:
It looks like currently we're ignoring the `/ln` flag on variables when parsing `.sym` files. Example below:
```
FormatVersion=6.0 // Do not edit this line!
Title="Untitled"

{SENDRECEIVE}

[FSM_ERRORS]
ID=040h	// This message contains a bit for each error that the FSM might throw
Len=8
Var=papps_out_of_range bit 0,1 /ln:"Primary Acceleration Pedal Position Sensor Out Of Valid Range"
```

Please let me know if you need any more info! Thanks!

eerimoq, 2019-01-25:
Feel free to implement it! =)

eerimoq, 2019-02-13:
By the way, `/ln` is not "Comment", it's "Long Name". `// ...` is the comment.

I'm working on several fixes and improvements to the SYM parser. This issue will likely be implemented.

garethellis0, 2019-02-13:
Ah, that makes a lot more sense. I've updated the issue title to reflect this. Thank you!

eerimoq, 2019-02-14:
I implemented comments, but not long name. I'll not implement long name as there is no obvious place for it in cantools database interface.

closed 2019-02-16 by eerimoq