cantools/cantools #108: Rounding issue for range checks in generated C code

thdhondt, 2019-02-01:
Hello,

I am currently using `cantools` to generate C source code for the following dbc file: [abs.zip](https://github.com/eerimoq/cantools/files/2821891/abs.zip).
A complete version of the dbc is available [here](http://www.bosch-motorsport.de/content/downloads/Products/31624027147.html#) in the _Documents_ section.

The generated C file contains:
```c
bool abs_mm5_10_tx1_ay1_is_in_range(uint16_t value)
{
    return (value <= 65534.06817557760953987369081u);
}
```

This gives the following error when compiling the code:
``` sh
rounding.c: In function ‘rounding_mm5_10_tx1_ay1_is_in_range’:
rounding.c:132:22: error: invalid suffix "u" on floating constant
     return (value <= 65534.06817557760953987369081u);
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

I believe this issue comes from a rounding error when applying the scaling factor to the minimum and maximum signal values stored in the dbc.

One way to solve this would be to apply the range check before encoding the signal, then the minimum and maximum values can be used directly from the dbc without scaling or offset.

Do you have any other idea on how to fix this issue?

Thanks in advance for your help :).

eerimoq, 2019-02-02:
I'm not sure what I want, but let's start by adding a flag to change the existing range check functions as your second alternative. It should be easy to change this behavior later if there is a use case.

eerimoq, 2019-02-07:
Closing as the problem has been fixed.

closed 2019-02-07 by eerimoq